projectchrono/chrono #258: Cannot build CASCADE module with OpenCASCADE 7.4.0 sourcecode

sprhawk, 2020-08-19:
Project Chrono revision: 2be212d4135d82c1879447af9d901b118a13ea1b
OpenCASCADE version: 7.4.0
Build Host: Debian 10 x64

OpenCASCADE build command line:
```
cmake -H. -BRelWithDebInfo -DCMAKE_INSTALL_PREFIX=$(dirname $PWD)/local -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUSE_FREEIMAGE=ON -DUSE_RAPIDJSON=ON -DUSE_TBB=ON -DUSE_VTK=ON -DBUILD_LIBBRARY_TYPE=Shared -DBUILD_SAMPLES_QT=OFF   ../opencascade-7.4.0/
```

Project Chrono build command line
```
cmake -H. -BRelWithDebInfo -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_INSTALL_PREFIX="$(dirname $PWD)/local"  -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_MODULE_IRRLICHT=ON -DENABLE_MODULE_POSTPROCESS=ON -DIRRLICHT_ROOT="$(dirname $PWD)/irrlicht-1.8.4/" -DENABLE_MODULE_CASCADE=ON -DCASCADE_ROOT="$(dirname $PWD)/opencascade-7.4.0" -DCASCADE_LIBDIR=$(dirname $PWD)/local/lib -DCASCADE_INCLUDE_DIR=$(dirname $PWD)/local/include/opencascade ../chrono/
```

Error Message:
```
chrono/src/chrono_cascade/ChCascadeMeshTools.cpp:20:10: fatal error: Handle_TopoDS_HShape.hxx: No such file or directory
 #include <Handle_TopoDS_HShape.hxx>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
chrono/src/chrono_cascade/ChCascadeDoc.cpp:23:10: fatal error: Handle_TopoDS_HShape.hxx: No such file or directory
 #include <Handle_TopoDS_HShape.hxx>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

I find in official documentation, this file `Handle_TopoDS_HShape.hxx` exists in occt-6.9.0, but not in occt-7.4.0, however the [CASCADE Module](http://api.projectchrono.org/module_cascade_installation.html) said chrono only be compatitle to occt-7.4.0

rserban, 2020-08-19:
The current **development** version of Chrono is compatible with OCC v7.4.0.  For that you should use the code in the `develop` branch of the Chrono GitHub repository.

I suspect you are using the release 5.01 of Chrono which is indeed compatible with OCC v6.9.0 (see [http://api.projectchrono.org/5.0.0/module_cascade_installation.html](http://api.projectchrono.org/5.0.0/module_cascade_installation.html))

sprhawk, 2020-08-19:
@rserban thanks

closed 2020-08-19 by sprhawk