projectchrono/chrono #260: Collision Detection: Distance of colliding objects depends on timestep

lippch, 2020-08-25:
Dear chrono-team,

I have encountered a problem when using collision detection with 2D-paths. I was hoping that you could tell me whether I am doing something wrong or if there is a problem in the collision detection algorithm.

The result from collision detection seems to be depending on the timestep used for calculation.
With a bigger timestep the calculated distance between the colliding objects increases linearly.
If the relative velocity between the colliding objects is ZERO, the distance shall be ZERO too. (When using the code below, this should be the case when the base arc is touching the follower -> shown in the screenshot)

Thank you in advance for your help!

[image]

You can reproduce the behaviour using the following code (based on demo_IRR_collision_2D.cpp)

`
// =============================================================================
// PROJECT CHRONO - http://projectchrono.org
//
// Copyright (c) 2014 projectchrono.org
// All rights reserved.
//
// Use of this source code is governed by a BSD-style license that can be found
// in the LICENSE file at the top level of the distribution and at
// http://projectchrono.org/license-chrono.txt.
//
// =============================================================================
// Authors: Alessandro Tasora
// =============================================================================
//
// Demo code about using paths for 2d collisions
//
// =============================================================================

#include "chrono/physics/ChBodyEasy.h"
#include "chrono/physics/ChLinkMotorRotationSpeed.h"
#include "chrono/physics/ChLinkTrajectory.h"
#include "chrono/physics/ChSystemNSC.h"
#include "chrono/core/ChGlobal.h"
#include "chrono/motion_functions/ChFunction.h"

#include "chrono_irrlicht/ChIrrApp.h"
#include "chrono/serialization/ChArchiveXML.h"
#include "chrono/serialization/ChArchiveBinary.h"

// Use the namespaces of Chrono
using namespace chrono;
using namespace chrono::geometry;
using namespace chrono::irrlicht;

// Use the main namespaces of Irrlicht
using namespace irr;
using namespace irr::core;
using namespace irr::scene;
using namespace irr::video;
using namespace irr::io;
using namespace irr::gui;

int main(int argc, char* argv[]) {
    GetLog() << "Copyright (c) 2017 projectchrono.org\nChrono version: " << CHRONO_VERSION << "\n\n";

    // Create a Chrono::Engine physical system
    ChSystemNSC mphysicalSystem;

    // Create the Irrlicht visualization (open the Irrlicht device,
    // bind a simple user interface, etc. etc.)
    ChIrrApp application(&mphysicalSystem, L"Paths", core::dimension2d<u32>(1200, 1000), false, true);

    // Easy shortcuts to add camera, lights, logo and sky in Irrlicht scene:
    application.AddTypicalLogo();
    //application.AddTypicalSky();
[... 273 lines cut ...]
        // Write file
        ChStreamOutBinaryFile mfileo(file.c_str());
        ChArchiveOutBinary marchiveout(mfileo);
        mphysicalSystem.ArchiveOUT(marchiveout);

        // Read file
        ChStreamInBinaryFile mfilei(file.c_str());
        ChArchiveInBinary marchivein(mfilei);

        mphysicalSystem.ArchiveIN(marchivein);
    }

    while (application.GetDevice()->run()) {
        application.BeginScene();

        application.DrawAll();

        application.DoStep();

        application.EndScene();
    }

    return 0;
}
`

tasora, 2020-08-28:
> CD-Distance_Timestep

Hi

sorry I have no time to compile your test code, but here is my guess: 
most likely, what you see is an artifact of the DVI nonsmooth 
formulation that we use. In fact, as we explained in some papers (ex.  
"A matrix-free cone complementarity approach for solving large-scale, 
nonsmooth, rigid body dynamics") we solve the *convexified* cone 
complementarity problem, that is equivalent to an optimization problem 
at the cost of allowing that the contact model is "associated", 
ie.contact speeds are in the polar cone of the friction cone, and thisw 
means that for nonzero friticon there is a 'departure' from the contact 
position. This is acceptable because the stabilization coefficient takes 
care of compensating this effect: after an initial departure, at steady 
state the two object will slide with a constant gap, of thickness  
g=h*u*v  proportional to time step h, friction coefficient u, tangent 
velocity v. So, the gap you see, is indeed proportional to the time 
step. Try using half friction, and probably the values for the gaps will 
be smaller. This, if there is some sliding tangential speed v, and you 
should live with it. But nor no *tangential* speed, indeed, the distance 
should be always zero (if this is your case, there is some issue).

regards

Alessandro Tasora

lippch, 2020-08-28:
Dear Mr. Tasora,

I have tried a few things and by setting the friction coefficients to zero the gap dissappears. So there seems to be no issue within the code.

Thank you very much for your response and the explanation!

Have a nice weekend!
Christian

closed 2020-08-28 by lippch