projectchrono/chrono #262: undefined reference to `chrono::irrlicht::ChIrrApp::ChIrrApp(chrono::ChSystem* ......

yumianhuli2, 2020-08-29:
Hello Chrono Team!
I have successfully installed the Irrlicht1.8.4 engine and run an example of irrlicht1.8.4
But when I compile Demos of Chrono Physics engine, the following error occurs below:

[image]

aluaces, 2020-08-30:
Hi, I think you have to install the GLUT developer libraries:

    sudo apt-get install freeglut3-dev

yumianhuli2, 2020-08-30:
> Hi, I think you have to install the GLUT developer libraries:
> 
> ```
> sudo apt-get install freeglut3-dev
> ```

Very help! Chrono works!
By the way, why are the vehicles in Chrono's vehicle demo running so slowly? Is there a  real-time demo?
Thank you very much!

Benatti1991, 2020-08-31:
They really shouldn't, are you building in Debug or Release?

yumianhuli2, 2020-08-31:
> They really shouldn't, are you building in Debug or Release?

Successfully compiled

closed 2020-09-04 by yumianhuli2