projectchrono/chrono #269: Installation error

h-pervez, 2020-10-01:
Hello all,

I am very new to programming/Python and currently facing this error when trying to install Pychrono using: 
conda install -c projectchrono pychrono in Anaconda 3 prompt.

I get the following message and not sure how to fix it, any help is greatly appreciated! Thanks :-)

--------------------------------------------------------------------------------------------------------------

(base) C:\>conda install -c projectchrono pychrono
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible so
lve.
Solving environment: failed with repodata from current_repodata.json, will retry
 with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible so
lve.
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abor-
failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pychrono -> python[version='>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']

Your python: python=3.8

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

Benatti1991, 2020-10-01:
Python 3.8 support has been added recently, so it is available only for the develop version.
Just type:
`conda install -c projectchrono/label/develop pychrono`
All the best,
Simone

h-pervez, 2020-10-02:
Thanks for your reply, I tried this but when running my script I get some further errors..I am stuck!

Benatti1991, 2020-10-02:
I cannot reproduce your error. Try using a conda environment (this is preferable in general)
```
conda create -n p38 python=3.8
activate p38
conda install -c projectchrono/label/develop pychrono
```

GiovanniBorre, 2020-12-18:
Hello all, I'm also stuck in something similar. Trying to install pychrono I tried to create a Conda environment
as you suggested, but I get this error:

`Collecting package metadata (current_repodata.json): done`
`Solving environment: failed with initial frozen solve. Retrying with flexible solve.`
`Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.`
`Collecting package metadata (repodata.json): done`
`Solving environment: failed with initial frozen solve. Retrying with flexible solve.`
`Solving environment: / `
`Found conflicts! Looking for incompatible packages.`
`This can take several minutes.  Press CTRL-C to abort.`
`failed                         `                                                 
`UnsatisfiableError: The following specifications were found to be incompatible with each other:`
`Output in format: Requested package -> Available versions`
 What should I do? Thanks!