projectchrono/chrono #271: Homebrew formula

zbeekman, 2020-10-12:
Hi,

I'm interested in creating a homebrew formula for chrono. I think I'll need to turn the demos off (since homebrew doesn't like app bundles or guis in homebrew-core formulae) but I wanted to ensure that:

1. I'm building a relatively full featured chrono
2. Idiosyncrasies/issues that may prevent addition to homebrew can be addressed

Concerning point 1 above: Right now I'm adding the following CMake flags to Homebrew's defaults:

```
-DBUILD_DEMOS=OFF
-DCMAKE_SHARED_LINKER_FLAGS=-lomp
-DENABLE_HDF5=ON
-DENABLE_MODULE_IRRLICHT=ON
-DENABLE_MODULE_POSTPROCESS=ON
-DENABLE_MODULE_VEHICLE=ON
-DENABLE_MODULE_PYTHON=ON
-DENABLE_OPENMP=ON
```

I tried to also enable OpenCascade but CMake failed to find the headers/libraries... this is fine, I can manually add them to enable OpenCascade as well, most likely. Are there any other features that I should be enabling?

My biggest questions are about parallelism: Obviously CUDA on macOS is more or less a non-starter, but is it possible to enable MPI based parallelism without thrust & blaze?

Concerning point 2 above: CMake seems to figure out where libomp.dylib is, but is not passing it to the linker when building shared libraries. This seems to me like it is a bug in the build system, however, I can find the brewed libomp and pass in `libomp.dylib` explicitly (or at least ensure that the linker search path has libomp in it and pass `-lomp`). Has anyone tried to look at this in the past and get the linker of macOS to find brewed libomp without additional input from the user?

zbeekman, 2020-10-12:
Just FYI, without specifying anything special beyond `-DENABLE_OPENMP=ON` CMake finds `libomp` but the build system does not pass it to the linker when building shared libraries:

```
 OpenMP_CXX_FLAGS                 -Xclang -fopenmp
 OpenMP_CXX_LIB_NAMES             libomp
 OpenMP_C_FLAGS                   -Xclang -fopenmp
 OpenMP_C_LIB_NAMES               libomp
 OpenMP_libomp_LIBRARY            /usr/local/lib/libomp.dylib
```

scpeters, 2022-10-01:
I am experimenting with a formula in the following tap: https://github.com/osrf/homebrew-simulation/pull/2068