projectchrono/chrono #272: Tutorial error: core/demo_CH_buildsystem.py

h-pervez, 2020-10-13:
I see the following error when trying to run this example:

  File "C:\Users\pervez\Anaconda3\Lib\site-packages\pychrono\demos\core\demo_CH_buildsystem.py", line 77, in <module>
    my_system.GetContactContainer().ReportAllContacts(my_rep)

  File "C:\Users\pervez\Anaconda3\lib\site-packages\pychrono\core.py", line 25440, in ReportAllContacts
    return _core.ChContactContainer_ReportAllContacts(self, callback)

RuntimeError: SWIG director method error. Error detected when calling 'ReportContactCallback.OnReportContact'

h-pervez, 2020-10-15:
I used a conda install on Windows 7, and I currently have Python 3.8.3 installed. How do I check if the demo is consistent with Pychrono? Thanks!

Benatti1991, 2020-10-15:
Hi,
they look consistent, the problem appears to be related to our conda package, while I cannot reproduce it when building from sources. We are working on fixing it an we'll keep you posted.
Simone

Benatti1991, 2020-10-19:
Hi, 
the problem you experienced was due to the conda packages being built with SWIG 3. I changed to SWIG 4 and the develop packages work fine. Please install the develop version number 688 and let us know if the problem is solved so that we can close the issue.
Simone

h-pervez, 2020-10-20:
Hello Simone, this definitely solved the issue, the demo now works fine! Thanks :D

closed 2020-10-20 by Benatti1991

aditya-shirwatkar, 2021-05-26:
@Benatti1991 @rserban I am facing the same issue.

[image]

`OS: Ubuntu 18.04`
`Python installation: conda install -c projectchrono/label/develop pychrono`
`Python Version: 3.7.10`
and I am running the demo python files on the current develop branch on GitHub

Thanks

aditya-shirwatkar, 2021-05-26:
> @Benatti1991 @rserban I am facing the same issue.
> 
> [image]
> 
> `OS: Ubuntu 18.04`
> `Python installation: conda install -c projectchrono/label/develop pychrono`
> `Python Version: 3.7.10`
> and I am running the demo python files on the current develop branch on GitHub
> 
> Thanks
> 
> Update: `Python installation: conda install -c projectchrono pychrono` also produces the same error

On further inspection I found that files are generated using SWIG 3.0.12
[image]

You mentioned to use version 688 but there is no such version on conda repo here https://anaconda.org/projectchrono/pychrono/files

Benatti1991, 2021-05-26:
@aditya-shirwatkar the version numbers got reset after last February's release (6.0). Try to uninstall and reinstall pychrono develop, you should get 223 version.

aditya-shirwatkar, 2021-05-26:
> @aditya-shirwatkar the version numbers got reset after last February's release (6.0). Try to uninstall and reinstall pychrono develop, you should get 223 version.

I observed this
* For linux-64
 [image]
* For windows-64
[image]

Maybe this issue needs a fix for linux-64 versions

Benatti1991, 2021-05-26:
Good point, I thought Linux was using SWIG4, I'm gonna fix it soon.

dummifiedme, 2022-08-05:
Was it fixed?
I facing the same error on the same tutorial problem.

```
Traceback (most recent call last):
  File "/home/user/Desktop/Compund Bow/pyChrono Trial/demo_CH_buildsystem.py", line 77, in <module>
    my_system.GetContactContainer().ReportAllContacts(my_rep)
  File "/home/user/anaconda3/envs/chrono/lib/python3.10/site-packages/pychrono/core.py", line 26667, in ReportAllContacts
    return _core.ChContactContainer_ReportAllContacts(self, callback)
RuntimeError: SWIG director method error. Error detected when calling 'ReportContactCallback.OnReportContact'

```

`OS: Manjaro Linux`
`Python Installation; conda install -c projectchrono pychrono
`Python Version: 3.10.5`

rserban, 2022-08-05:
The conda packages are being worked on. We hope to have them ready shortly.
In the meantime, you can try to build PyChrono from sources.  Everything works fine in that case.

dummifiedme, 2022-08-05:
> In the meantime, you can try to build PyChrono from sources. Everything works fine in that case.

Just to make sure, we don't need anaconda if we build from source? (I am asking about it since I am a little low on root storage)