projectchrono/chrono #273: About vehicle of pure numerical calculation

yumianhuli2, 2020-10-15:
Is there an example of a chrono vehicle model that contains no Irrlicht visualizations but only pure numerical calculation?Thanks！

yumianhuli2, 2020-10-16:
> In general, it's pretty straightforward to modify any of the existing vehicle demos to eliminate the rendering. Of course, such a demo cannot control the vehicle in an interactive manner (i.e. through keyboard inputs); you'd need to use driver inputs from a file (ChDataDriver) or some closed-loop drvier controller (e.g. ChPathFollowerDriver). Except for this, all you need to do is remove everything having to do with the Irrlicht visualization "app" (ChWheeledVehicleIrrApp) and, as needed, change the exit condition for the simulation loop (for example, based on a final simulation time).
> 
> An existing demo that illustrates this is demo_VEH_Shock (https://github.com/projectchrono/chrono/blob/develop/src/demos/vehicle/demo_ISO2631/demo_VEH_Shock.cpp). Simply comment out line 47:
> #define USE_IRRLICHT

Thanks!@reserban
More Question:
// Simulation step size (should not be too high!)
double step_size = 1e-3;
With this step size, the vehicle movement is very slow. I don't know what is the conversion between this and real time?

If I do the simulation in steps of 0.1 or 1, the simulation is very fast!
Please tell me how to do it for real time

rserban, 2020-10-16:
Chrono does not guarantee real-time simulation.  However, certain models (including some vehicle models) can run in real-time on commodity hardware.

You must set the solver parameters, including the integration step size, to ensure a stable simulation.  Once you find the largest step size that produces a stable and sufficiently accurate simulation, you can check how close you are to real-time simulation by comparing the simulated time with the elapsed wall clock time.

For a Chrono::Vehicle model, I can confidently say that a step size of 0.1 (let alone 1) is much too large.  If you were to visualize your simulation (which I suspect you disabled) and/or check the solution (for example by looking at how well the joint constraints are satisfied) you will most likely find that the results you get with such time step sizes are useless.

----

On a different note, if you have not already done so, I recommend you subscribe to the Chrono user mailing list (https://groups.google.com/g/projectchrono).  You can browse past conversations (some of which discuss topics related to your question) and post this type of questions there.

yumianhuli2, 2020-10-16:
Thanks！This should be a vvery good real time example
https://www.youtube.com/watch?v=urWbpTDDzpk&list=PLdUBkWvqFrxCQhzv2YTw6uuyg4x5NdQbJ&index=13