projectchrono/chrono #275: Unable to visualize demo_FEA_contacts.py in pychrono (black screen only with pychrono logo)

h-pervez, 2020-10-26:
Hello, when trying to run the aforementioned demo, I just see a black screen with the Pychrono logo, could anybody please help?
Thanks in advance!

Benatti1991, 2020-10-26:
Hi,
sounds odd, it works fine for me. It might be a data path error. Make sure you are launching it in its own directory.
Like 
`cd C:\Users\your_name\Anaconda3\envs\some_env\Lib\site-packages\pychrono\demos\fea
python demo_FEA_contacts.py`
Simone

h-pervez, 2020-10-27:
Thanks I rechecked the path and it looks fine:

runfile('C:/Users/pervez/Anaconda3/Lib/site-packages/pychrono/demos/fea/demo_FEA_contacts.py', wdir='C:/Users/pervez/Anaconda3/Lib/site-packages/pychrono/demos/fea')
Copyright (c) 2017 projectchrono.org

[image]

uebian, 2025-11-28:
Are you still encountering this issue with the latest PyChrono?

rserban, 2025-11-28:
The code has gone through 5 new versions since this issue was posted.

closed 2025-11-28 by rserban