projectchrono/chrono #277: PID controller in PyChrono

mcauwelier, 2020-11-03:
Hi,

I would like to use a PID controller in PyChrono as described in the demo (demo_IRR_motors.cpp). The demo recommends inheriting from the ChFunction_SetpointCallback base class, but pychrono has no class ChFunction_SetpointCallback. Is it not yet added to the python wrapper (I am using the latest development branch, compiled from source)? 

If so, what is the alternative, just write a custom Force/Torque Function? What are the the downsides of this method as hinted at in the demo file?

Thank you in advance,

Best regards,

Matthias

Benatti1991, 2020-11-04:
Hi Matthias,
I wrapped ChFunction_SetpointCallback, I'm also going to add a Python version of demo_IRR_motors and let you know when I'm done.
Simone

mcauwelier, 2020-11-04:
Hi Simone,

Thank you for your quick response and effort, will try it out.

Best regards,
Matthias

closed 2020-11-04 by mcauwelier