projectchrono/chrono #280: Infinite recursion in the assignment operator of ChNodeFEAxyz

ranjeethmahankali, 2020-11-10:
The following snippet recreates the problem.
```
ChNodeFEAxyz node1(ChVector<>(42.0, 42.0, 42.0));
ChNodeFEAxyz node2(ChVector<>(0.0, 0.0, 0.0)); 
node2 = node1; // Fails.
```
The last line causes infinite recursion. I found the problem in the source code. I also verified the recursion by building the source with debug information and stepping through it. It seems like a simple fix on the surface but I don't know enough about all the possible side effects to be able to fix the issue myself.

We are still using version 4.0, because 5.0 proved unstable (corrupted memory causing access violation errors). Can this fix be cherry-picked into 4.0?

rserban, 2020-11-10:
I see the problem in ChNodeFEAxyz::operator=().  Most likely a copy-and-paste error...  I'll fix this and push.

I can also create a hotfix for this in 4.0, but I'd like to hear more about the access violation errors in 5.0 you discovered.  Could you please give some more details?

ranjeethmahankali, 2020-11-10:
Thanks for the quick response and the fix!

I was getting the access violation error when running this example: http://api.projectchrono.org/tutorial_demo__f_e_a_ancf_shell.html

It often happened inside the for loop that creates the elements, when calling the `AddLayer` function. But it was not consistent, sometimes it would occur in the first iteration of the loop and sometimes in the 7th or 10th iteration. Sometimes it also happened in other function calls (but still inside the same loop).

rserban, 2020-11-12:
I pushed a fix for the ChNodeFEAxyz bug in the 'develop' branch.   I'll look into hot fixes for releases v4 and v5, but first I'd like to understand better the access violation issue you reported with the code in v5.

I am not able to reproduce this problem with the latest code in 'develop'.   Before I investigate this with the v5 code, would it be possible for you to test the code in the 'develop' branch?  Do you see the same problem with demo_FEA_ANCFshell?

Also, please provide some details on your OS, compiler, etc.

ranjeethmahankali, 2020-11-14:
I was getting the access violation error in the `main` functions of several FEA tutorials. I was also able to recreate this error on multiple machines (personal and work PCs). This was almost a month ago now.
Yesterday and today I've been trying to recreate the error but I can't. I tried everything that I was trying back then but still no error.

I am on Windows. Both times, I compiled the latest master, so maybe its already fixed? I also updated my VS2019 and MSVC during the last month.

Not sure how I can recreate the problem now. Let me know if there is anything else I can help with and thanks for the fix!

rserban, 2020-11-14:
The latest code on the 'master' branch is release 5.01.   This is the code you said gave you that run-time error.
The very latest code (now about 9 months ahead of the last release) is on the 'develop' branch.  This is what we recommend users to work off, unless there are other compelling reasons to stick with a release version.

rserban, 2020-11-14:
So my question was: Which version of Chrono did you just test?  The one on 'master' or on 'develop'?

ranjeethmahankali, 2020-11-15:
Oh sorry, didn't think there was a question in there.
Both times I just pulled the latest 'master'. Last time I was getting the error. This time I am not getting the error.

ranjeethmahankali, 2020-11-17:
@rserban Just an update on the access violation error:
Once we started using the latest develop, we were getting the access violation error again, only in debug mode and not in release mode. I carefully compared the project files of the demos with our application, trying out one setting at a time to see if it fixes the issue.

When I set the "Enable Enhanced Instruction Set" option to `Advanced Vector Extensions 2 (/arch:AVX2)` the error goes away. So that means its not possible to use chrono without supporting this instruction set. Is this intended?

rserban, 2020-11-17:
Is this still with demo_FEA_ancfShell?

In any case, if supported, /arch:AVX2 should be enabled by default.  Could you please check the output from CMake during configuration (look for "Searching for AVX...")?   Do you see something like:
```
Searching for AVX...
   Found AVX 2.0 extensions, using flags:  /arch:AVX2
```

Please also check whether or not the generated ChConfig.h (in the build tree, under chrono/) contains lines such as:
```
#define CHRONO_HAS_AVX
#define CHRONO_AVX_LEVEL "2.0"
#define CHRONO_AVX_1_0
#define CHRONO_AVX_2_0
```

Finally, please provide your CMakeOutput.log.

ranjeethmahankali, 2020-11-17:
It is with the demo: http://api.projectchrono.org/tutorial_demo__f_e_a_shells.html

The chrono project files were generated correctly. So nothing is wrong with cmake. I had to enable avx2 in our application's project file. This project file has nothing to do with your cmake.

ranjeethmahankali, 2020-11-18:
This is what I see in the CMake output:
```
Searching for AVX...
Performing Test DETECTED_AVX_20
Performing Test DETECTED_AVX_20 - Success
Performing Test DETECTED_AVX_10
Performing Test DETECTED_AVX_10 - Success
  Found AVX 2.0 extensions, using flags:  /arch:AVX2
Searching for FMA...
Performing Test DETECTED_FMA
Performing Test DETECTED_FMA - Success
  Found FMA extensions, using flags:  /arch:AVX2
```
The `CMakeOutput.log` is attached here:
[CMakeOutput.log](https://github.com/projectchrono/chrono/files/5562782/CMakeOutput.log)

And I do see the following lines in the `ChConfig.h` file:
```
#define CHRONO_HAS_AVX
#define CHRONO_AVX_LEVEL "2.0"
#define CHRONO_AVX_1_0
#define CHRONO_AVX_2_0
```
Like I said before, the problem goes away if we enable the avx2 instruction in our own application, even though we are only linking to chrono dynamically at runtime.
Unfortunately, enabling avx2 in our own application causes unintended side effects elsewhere so we can't enable it.  Can you help us figure out another solution for this?

ranjeethmahankali, 2020-11-19:
Ok, disabling AVX2 when building chrono seems to solve all problems. I changed this in the project file, and commented out the #defines in the ChConfig.h

Is there anything else I have to do to properly disable the avx2?

rserban, 2020-11-20:
Another user requested the ability to optionally disable SIMD support at configuration time, so I will implement this in the near future.   Note that I plan on providing a single CMake option that could be used to disable all SIMD support (including AVX2).  Will that work in your case?

ranjeethmahankali, 2020-11-20:
Yes, that works. Thanks!

commit 6439ce54 (2020-11-21): Add CMake option (USE_SIMD) to disable SIMD support.

Fixes #280

(cherry picked from commit c571b5d093fcf8c3470f967ad82b1ec071b1c8fe)
closed 2020-11-21 by rserban