projectchrono/chrono #281: Bug: demo_CAS_robot.py fails to load step files

blooop, 2020-11-11:
Hi, 

I have installed via conda forge on python 3.6.

I am trying to run the demo:
`https://github.com/projectchrono/chrono/blob/develop/src/demos/python/cascade/demo_CAS_robot.py`

However it prints a load of error messages and only displays a floating circle.

It looks like something has gone wrong in the step file/loading code.  

[image]

```
Example: Load a STEP file, generated by a CAD.
Please wait! this may take a while to load the file...
  ***  Error on Record 771 (on 164929 -> 0 % in File)  ***  Ident #465
Complex Type incorrect : SOLID_ANGLE_UNIT / SI_UNIT ... 
  ***  Error on Record 1925 (on 164929 -> 1 % in File)  ***  Ident #1096
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 1927 (on 164929 -> 1 % in File)  ***  Ident #1096
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2204 (on 164929 -> 1 % in File)  ***  Ident #1253
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2206 (on 164929 -> 1 % in File)  ***  Ident #1253
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2230 (on 164929 -> 1 % in File)  ***  Ident #1259
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2232 (on 164929 -> 1 % in File)  ***  Ident #1259
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2265 (on 164929 -> 1 % in File)  ***  Ident #1274
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2267 (on 164929 -> 1 % in File)  ***  Ident #1274
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2363 (on 164929 -> 1 % in File)  ***  Ident #1343
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2365 (on 164929 -> 1 % in File)  ***  Ident #1343
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2401 (on 164929 -> 1 % in File)  ***  Ident #1361
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2403 (on 164929 -> 1 % in File)  ***  Ident #1361
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2463 (on 164929 -> 1 % in File)  ***  Ident #1388
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2465 (on 164929 -> 1 % in File)  ***  Ident #1388
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2486 (on 164929 -> 1 % in File)  ***  Ident #1391
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2488 (on 164929 -> 1 % in File)  ***  Ident #1391
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 2797 (on 164929 -> 1 % in File)  ***  Ident #1585
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 2799 (on 164929 -> 1 % in File)  ***  Ident #1585
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 3329 (on 164929 -> 2 % in File)  ***  Ident #1955
Complex Type incorrect : NAMED_UNIT / LENGTH_UNIT ... 
  ***  Error on Record 29641 (on 164929 -> 17 % in File)  ***  Ident #16337
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 29643 (on 164929 -> 17 % in File)  ***  Ident #16337
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 29766 (on 164929 -> 18 % in File)  ***  Ident #16442
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 29768 (on 164929 -> 18 % in File)  ***  Ident #16442
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 29834 (on 164929 -> 18 % in File)  ***  Ident #16490
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
  ***  Error on Record 29836 (on 164929 -> 18 % in File)  ***  Ident #16490
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 29920 (on 164929 -> 18 % in File)  ***  Ident #16538
Complex Type incorrect : RATIONAL_B_SPLINE_SURFACE / BOUNDED_SURFACE ... 
[... 45 lines cut ...]
Complex Type incorrect : REPRESENTATION_ITEM / GEOMETRIC_REPRESENTATION_ITEM ... 
  ***  Error on Record 110368 (on 164929 -> 66 % in File)  ***  Ident #68886
Complex Type incorrect : GEOMETRIC_REPRESENTATION_ITEM / CURVE ... 
Report : 27 unknown entities
Irrlicht Engine version 1.8.4
Linux 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020 x86_64
Using plain X visual
This driver is not available in Linux. Try OpenGL or Software renderer.
Cannot use default video driver - fall back to OpenGL 
Irrlicht Engine version 1.8.4
Linux 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020 x86_64
Using renderer: OpenGL 4.6.0
Quadro P3200/PCIe/SSE2: NVIDIA Corporation
OpenGL driver version is 1.2 or better.
GLSL version: 4.6
Loaded texture: /anaconda3/envs/pychrono36/share/chrono/data/fonts/arial80.bmp
Loaded texture: /anaconda3/envs/pychrono36/share/chrono/data/skybox/sky_lf.jpg
Loaded texture: /anaconda3/envs/pychrono36/share/chrono/data/skybox/sky_dn.jpg
Loaded texture: /anaconda3/envs/pychrono36/share/chrono/data/skybox/sky_up.jpg
Loaded texture: /anaconda3/envs/pychrono36/share/chrono/data/logo_pychrono_alpha.png
Could not open file of texture: XEFFECTS_SM_512
XEffects: Please ignore previous warning, it is harmless.
Could not open file of texture: XEFFECTS_SM_512_2
XEffects: Please ignore previous warning, it is harmless.
```

Benatti1991, 2020-11-11:
Hi blooop,
via conda forge? It sounds strange since conda forge package shouldn't have Irrlicht module.
The conda forge package is not provided by us, our package in is our anaconda cloud channel :
https://anaconda.org/projectchrono/pychrono
Install this package (I suggest develop version) and let's see what happens.
Simone

blooop, 2020-11-11:
Sorry, that was a typo.  These are the commands I ran.

```
sudo apt install libirrlicht1.8
conda install -c projectchrono pychrono
conda install -c neok.m4700 glibc #required on ubuntu 18.04 has glibc 2.27 and glibc2.29 is required
conda install -c dlr-sc pythonocc-core #otherwise import OCC fails
```

blooop, 2020-11-11:
It works!

Thanks for your help!

For posterity:
on python 3.7:
```
sudo apt install libirrlicht1.8
conda activate pychrono37env #change for your env name
conda install -c projectchrono/label/develop pychrono #use dev version to get working examples
conda install -c dlr-sc pythonocc-core #otherwise import OCC fails
```
the example works as expected.

closed 2020-11-11 by blooop