projectchrono/chrono #287: Library 'ChronoEngine' in package Chrono is not installed properly

zhangtianjia, 2021-01-05:
Dear developers,
    When I run `cmake` command with a CMakeList which set `Chrono_DIR` to link to the precompiled chrono directory, the Terminal throw out:
```
Library 'ChronoEngine' in package Chrono is not installed properly
Library 'ChronoEngine_irrlicht' in package Chrono is not installed properly
--    CHRONO_CXX_FLAGS         (Compiler flags):  -std:c++14  -openmp  /arch:SSE2  /arch:AVX2  /arch:AVX2  /Zc:__cplusplus  /wd4275
--    CHRONO_C_FLAGS           (Compiler flags):   -openmp  /arch:SSE2  /arch:AVX2  /arch:AVX2   /wd4275
```
Then I run `make`, it shows error:
```
c++: error: /arch:SSE2: No such file or directory
c++: error: /arch:AVX2: No such file or directory
c++: error: /arch:AVX2: No such file or directory
c++: error: /Zc:__cplusplus: No such file or directory
c++: error: /wd4275: No such file or directory
c++: error: unrecognized command line option ‘-std:c++14’; did you mean ‘-std=c++14’?
CMakeFiles/scenenet_room_generator.dir/build.make:81: recipe for target 'CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o' failed
make[2]: *** [CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/scenenet_room_generator.dir/all' failed
make[1]: *** [CMakeFiles/scenenet_room_generator.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2
```
Could you please help me find where the  problem is?

In the `CMakeList.txt`, I write the following codes to link to chrono:
```
set(Chrono_DIR "../../My_stereo/ProjectChrono/cmake")
find_package(GSL REQUIRED)
find_package(Chrono
             COMPONENTS Irrlicht
             CONFIG)
if (NOT Chrono_FOUND)
  message("Could not find Chrono or one of its required modules")
  return()
endif()
```
where the path is linked to the `cmake` directory containing the `ChronoConfig.cmake` file.

closed 2021-01-05 by zhangtianjia

GG1272, 2023-09-28:
I have encountered the same problem. Have you resolved it?

rserban, 2023-09-29:
@GG1272 - please provide more details on what OS, compiler, and Chrono version you use. 

Also, when you say that you "encountered the same problem" what do you mean? That you also end up with an unrecognized compiler flag `-std:c++14`? 

We never found out from @zhangtianjia how he ended up with that since they closed the issue without providing any details...
By the way, you may want to use the latest Chrono code in the `main` branch. There were some changes on compiler settings.