projectchrono/chrono #288: Errors in "demo_CH_buildsystem.py"

likeLyl, 2021-01-19:
Hi all,

I'm new to use Chrono. I try to use pychrono to running some demos.
Then, I got two errors when I running "demo_CH_buildsystem.py" this file. However, I solved the first error but I still talk about this because I'm not sure if it is a correct way.

**FIRST ERROR:**
```
        _Second tutorial: create and populate a physical system
        Traceback (most recent call last):
          File "E:/Program Files (x86)/python_with_chrono/Core/demo_CH_buildsystem.py", line 36, in <module>
            bodyA.GetCollisionModel().AddBox(material, 10, 1, 10)
          File "E:\Program Files (x86)\chrono_build1\bin\Release\pychrono\core.py", line 6563, in AddBox
            return _core.ChCollisionModel_AddBox(self, *args)
        TypeError: Wrong number or type of arguments for overloaded function 'ChCollisionModel_AddBox'.
          Possible C/C++ prototypes are:
            chrono::collision::ChCollisionModel::AddBox(double,double,double,chrono::ChVector< > const &,chrono::ChMatrix33< > const &)
            chrono::collision::ChCollisionModel::AddBox(double,double,double,chrono::ChVector< > const &)
            chrono::collision::ChCollisionModel::AddBox(double,double,double)_
```

I got this error when I running the following code:
`bodyA.GetCollisionModel().AddBox(material, 10, 1, 10)`

**My solution:**
After I check the C++ API and some examples, I change the code to :
```
bodyA.SetMaterialSurface(material)
bodyA.GetCollisionModel().AddBox(10, 1, 10)
```
And the error is missed. 

I found the other similar errors that all about "material" in Chrono, such as something like the above that need to set material. The solve is the same, call "variable.SetMaterialSurface(material)" firstly and remove the “material” parameter in error function.

**SECOND ERROR:**
```
Traceback (most recent call last):
  File "E:/Program Files (x86)/python_with_chrono/Core/demo_CH_buildsystem.py", line 124, in <module>
    for alink in my_system.Get_linklist():
TypeError: 'SwigPyObject' object is not iterable
```

The error code is 
```
for alink in my_system.Get_linklist():
    print ('  link: ', alink.GetName() )
```

it seems like  my_system.Get_linklist() cannot be iterable. I printed the length of the "my_system.Get_linklist()" and I got zero. In this file, one link is added in the "my_system". So it should output one but not.

Except for solving the first error, I haven't change any other codes.

I cannot figure out this error, Does anyone know how to solve this problem?

I will very appreciate it if you could give me any suggestion.

rserban, 2021-01-19:
Looks like you try to run PyChrono demos that are incompatible with the PyChrono version you have installed.   Please see the warning at the top of this documentation page: http://api.projectchrono.org/tutorial_table_of_content_pychrono.html

likeLyl, 2021-01-19:
> Looks like you try to run PyChrono demos that are incompatible with the PyChrono version you have installed. Please see the warning at the top of this documentation page: http://api.projectchrono.org/tutorial_table_of_content_pychrono.html

Yes! That's the reason! Thank you very much!
There is nothing wrong when I running the correct demo file.

closed 2021-01-19 by likeLyl