projectchrono/chrono #289: fail to compile the Chrono of develop version in Windows 10

likeLyl, 2021-01-20:
Hi all, I faced a critical problem when I compile develop Chrono Project in VS 2017.

I have built the master version and it is succeeded, but it failed to build CASCADE module with OpenCASCADE 7.4.0, the reason is https://github.com/projectchrono/chrono/issues/258. Unfortunately, VS2017 cannot build OpenCASCADE 6.9.0, so that I tried to build develop version from scratch.

I followed the same instruction (http://api.projectchrono.org/5.0.0/tutorial_install_chrono.html) to build develop version, which is failed to build any modules because the folder "release" and "debug" are empty after I compiled, also the VS2017 remind there are many failures.

This is the ERROR I faced:
[image]
I tried to build POSTPROCESS and IRRLICHT and other default modules only.

Below is the version of some libraries and tools I used.
system: Windows 10
Cmake: 3.19.2
irrlicht: 1.8.4
Eigen: 3.3.9
swigwin: 4.0.2

By the way, only one thing is different between I build two versions. I found that EIGEN3_INCLUDE_DIR in develop version always disappeared without error about Eigen in Cmake after I configured, then, it can be generated normally. I checked the CMakeCache.txt in two build folders, and I found the different about EIGEN3_INCLUDE_DIR , as shown below:

develop version:
[image]

master version:
[image]

I'm stuck on this problem all days, I will very appreciate it if you could give me any help.

One more thing, it also failed to build develop version after I deleted the master version.

dariomangoni, 2021-01-24:
Did you try, in CMake, to just go to File>Delete Cache and start back again?

likeLyl, 2021-01-24:
> Did you try, in CMake, to just go to File>Delete Cache and start back again?

Hi, Thank you for your reply very much.

I tried this but not works. So I delete the Chrono_develop and Chrono_buiild and start back again. Then, the problem with the Eigen library I mentioned above still exists.

However, I succeeded built the Chrono develop version, but I cannot import cascade from pychrono.
The error as shown below:

```
E:\Anaconda3\python.exe "E:/Program Files (x86)/test_cascade/demo_CAS_cascade.py"
Example: create OpenCascade shapes and use them as rigid bodies
Traceback (most recent call last):
  File "E:/Program Files (x86)/test_cascade/demo_CAS_cascade.py", line 16, in <module>
    import pychrono.cascade as cascade
  File "E:\Program Files (x86)\chrono_build\bin\Release\pychrono\cascade.py", line 13, in <module>
    from . import _cascade
ImportError: cannot import name '_cascade' from 'pychrono' (E:\Program Files (x86)\chrono_build\bin\Release\pychrono\__init__.py)
```
I have no idea why it happened. Do you know the cause of the problem? Many thanks.

nehelmalhotra, 2021-11-07:
> ImportError: cannot import name '_cascade' from 'pychrono' (E:\Program Files (x86)\chrono_build\bin\Release\pychrono\__init__.py)

Hey! I am running into the same problem. Were you able to resolve it?

likeLyl, 2021-11-17:
> > ImportError: cannot import name '_cascade' from 'pychrono' (E:\Program Files (x86)\chrono_build\bin\Release\pychrono__init__.py)
> 
> Hey! I am running into the same problem. Were you able to resolve it?

Hi, sorry I didn't solve this, so I change to use Linux to run Chrono and everything is alright!

nehelmalhotra, 2021-11-17:
Thanks for your reply! One of my colleagues might have figured out a way to compile this on Windows. I'll give it a shot and post here in case it works.

Alternatively, I will try to compile it on Linux as you have suggested!