projectchrono/chrono #292: 6.0.0 installs empty directory include/chrono_thirdparty/flatbuffers

yurivict, 2021-02-10:


JayTaves, 2021-02-10:
Hello, what operating system was this on? Which modules did you have enabled?

yurivict, 2021-02-10:
FreeBSD 12.2

cmake was invoked with these arguments:
```
-DBUILD_DEMOS:BOOL=false -DENABLE_HDF5:BOOL=true -DCMAKE_C_COMPILER:STRING="cc"  -DCMAKE_CXX_COMPILER:STRING="c++"  -DCMAKE_C_FLAGS:STRING="-O2 -pipe -fno-omit-frame-pointer  -fstack-protector-strong -fno-strict-aliasing "  -DCMAKE_C_FLAGS_DEBUG:STRING="-O2 -pipe -fno-omit-frame-pointer  -fstack-protector-strong -fno-strict-aliasing "  -DCMAKE_C_FLAGS_RELEASE:STRING="-O2 -pipe -fno-omit-frame-pointer  -fstack-protector-strong -fno-strict-aliasing "  -DCMAKE_CXX_FLAGS:STRING="-O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  "  -DCMAKE_CXX_FLAGS_DEBUG:STRING="-O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  "  -DCMAKE_CXX_FLAGS_RELEASE:STRING="-O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  "  -DCMAKE_EXE_LINKER_FLAGS:STRING=" -Wl,-rpath=/usr/local/lib/gcc10  -L/usr/local/lib/gcc10 -B/usr/local/bin -fstack-protector-strong "  -DCMAKE_MODULE_LINKER_FLAGS:STRING=" -Wl,-rpath=/usr/local/lib/gcc10  -L/usr/local/lib/gcc10 -B/usr/local/bin -fstack-protector-strong "  -DCMAKE_SHARED_LINKER_FLAGS:STRING=" -Wl,-rpath=/usr/local/lib/gcc10  -L/usr/local/lib/gcc10 -B/usr/local/bin -fstack-protector-strong "  -DCMAKE_INSTALL_PREFIX:PATH="/usr/local"  -DCMAKE_BUILD_TYPE:STRING="Release"  -DTHREADS_HAVE_PTHREAD_ARG:BOOL=YES  -DCMAKE_INSTALL_RPATH_USE_LINK_PATH=YES  -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON
```

yurivict, 2021-02-11:
Another empty directory is ```include/chrono_synchrono/flatbuffer/fbs``` when ```SYNCHRONO=ON```.

rserban, 2021-02-11:
I just amended the 6.0 release to fix a couple of issues, including this.  Unless I missed something, Chrono should not install any empty directories anymore.

NOTE:  I didn't retag 6.0.0 quite the right way, so if you have it locally cloned, you'll need to delete the local 6.0..0 tag and then fetch.

closed 2021-02-12 by rserban