projectchrono/chrono #293: Please explain in the README how to run tests

yurivict, 2021-02-10:
Building with ```BUILD_TESTING=ON``` and running ```gmake test``` or ```ctest``` doesn't do it, so it requires to be documented how to do it.

rserban, 2021-02-11:
We use `ctest` to run all unit tests, but this command must be issued in the top-level directory of the build tree.
I can add a note about this in the Readme file, but first could you please confirm that you can run them like above?

rserban, 2021-02-11:
You may actually be talking about batch-running the unit tests from an installation of Chrono?   If that is the case, I don't know of a straight-forward way of doing this.   

Anyway, I added a README file in the src/tests/ directory about running tests in the build and install trees.

closed 2021-11-06 by rserban