projectchrono/chrono #300: ChronoConfig.cmake isn't installed in correct folder

mablanchard, 2021-02-16:
Building Chrono 4.0.0, the CMake config. script get installed as `<prefix>/<libdir>/cmake/ChronoConfig.cmake`. This is somehow unfortunate because this is not a location CMake's [`find_package()` search procedure](https://cmake.org/cmake/help/latest/command/find_package.html?highlight=find_package#search-procedure) would search for `ChronoConfig.cmake`, leading to errors of the kind:

> -- Could NOT find Chrono (missing: Chrono_DIR)

I believe a better location would be `<prefix>/<libdir>/cmake/Chrono/ChronoConfig.cmake` which would allow something like the snippet below to just work:

```cmake
find_package(
  Chrono CONFIG
  HINTS
    "${prefix}")

```

mablanchard, 2021-02-16:
A workaround would be something like:

```cmake
set(Chrono_DIR "${prefix}/lib/cmake")
find_package(
  Chrono CONFIG)
````

scpeters, 2022-10-01:
fix in https://github.com/projectchrono/chrono/pull/421

PR #421 (2023-01-05): Fix cmake config install folder on Unix systems
As noted in #300, for Unix systems Chrono's cmake config file is installed to `<prefix>/lib/cmake/`, but this is not a [default search location for cmake's find_package function](https://cmake.org/cmake/help/latest/command/find_package.html#config-mode-search-procedure). This changes the install path to `<prefix>/lib/cmake/Chrono`, which is one of the default search paths.

Fixes #300.
commit 2e633c62 (2023-01-05): Fix cmake config install folder

Fixes #300.