projectchrono/chrono #307: Irrlicht GUI in PyChrono

SoWizard, 2021-03-31:
Running a compiled version of the C++ demo `demo_IRR_collisionNSC` allows the user to access the iirlicht GUI by pressing `i`. Users can also pause with space and rotate the visualisation by clicking and dragging the left or right hand mouse buttons.
[image]

Running the same example through PyChrono, the user can only use mouse-related actions (rotations, zooming). Any keystrokes are ignored (like pressing `i`). Keystrokes are sent to the previously opened application (Terminal, PyCharm, Finder, etc.).
[image]

Could keystrokes not be bound in PyChrono? If so, is there any way to enable it?

This is from the latest PyChrono release on Anaconda.
Both examples running on macOS 11.2.3.

SoWizard, 2021-03-31:
I setup PyChrono in a Windows 10 VM and it seems to work fine... Seems to be a macOS-related issue. 

[image]

Note, in the above post PyChrono is just installed through `conda` and iirlicht through Homebrew.

The C++ compiled version of `demo_IRR_collisionNSC` was of my own compilation through CMake.

Strangely the Windows 10 VM running the Python script has much better CPU Solver Time performance (1-3ms) compared to running the natively compiled C++ version on macOS. Anyway, not sure what should be done about the GUI not working on macOS.

rserban, 2021-03-31:
I don't have access to a Mac, but I suspect the performance difference you mention is due to Irrlicht rendering.
If you could also configure and build the C++ Chrono libraries on your Windows VM, it would be interesting to see what results you get when running some of the benchmark tests (you need to enable them during CMake configuration); these tests do not use run-time visualization.

SoWizard, 2021-04-06:
That would make sense - given how there's something else off about irrlicht. I haven't got the time at the moment but I might circle back in the future and try your suggestion.

closed 2021-11-07 by rserban