projectchrono/chrono #327: Application Error 0xc0000142 Windows 10 precompiled binaries

BeinerF, 2021-06-25:
Hi,
after installing Project Chrono 6.0.0 binaries for windows under windows 10 and
executing some of the *.exe files in /bin  I get without exception an application error 0xc0000142 .

Has anybody else experienced this ?
Cheers

NateRoiger, 2021-09-30:
I'm hitting the same error on both the binaries and the source. I posted to the [forums](https://groups.google.com/g/projectchrono/c/gdaoHoR_IOw/m/iuPmSlQVBAAJ?utm_medium=email&utm_source=footer) as well; still waiting for a reply.

rserban, 2021-09-30:
Nate, I replied to your forum post.  I don't think this is a Chrono issue, but rather a Windows installation/configuration issue.

NateRoiger, 2021-10-01:
I posted in the forum post - the issue I had was ChronoEngine.dll building with AVX2 instruction set, while my processor only supports AVX. Not sure if this is an issue with cmake or maybe the configuration. But forcing AVX then rebuilding the engine from source resolved the issue for me.

NateRoiger, 2021-10-04:
@rserban Please find your requested files attached. As for processors, it's AMD FX-8350. CPU-Z confirms AVX (but no AVX2)
[CMakeError.log](https://github.com/projectchrono/chrono/files/7281718/CMakeError.log)
[CMakeOutput.log](https://github.com/projectchrono/chrono/files/7281719/CMakeOutput.log)
[CMakeCache.txt](https://github.com/projectchrono/chrono/files/7281720/CMakeCache.txt)

rserban, 2021-10-05:
Nate, looking at the files you provided, it seems that the Chrono cmake configuration scripts correctly identified that AVX2 is not supported on your processor.  However, the subsequent test for FMA support (incorrectly) added the arch:AVX2 flag automatically.  I pushed some changes to the configuration files in the `feature/cmake_config` branch.  Could you please check out that branch, do a fresh cmake configuration (to make sure no cached variables are used), and let me know if this fixes your issue?   Thanks.

NateRoiger, 2021-10-05:
Radu, Everything looks resolved. Thank you!