projectchrono/chrono #328: Collision family of RigidTerrain patches not initialized correctly if having only one patch

chfeller, 2021-07-14:
Hi guys! 

To me this appears to be a bug, but it might well be that I'm misunderstanding something. In RigidTire::Initialize, the collision family of the terrain is only handed down to the individual patches if there is **strictly more** than one patch, see below.  Consequently, if there is only one patch in the terrain, this one will stick at the default collision family 0. I would have expected a '<=' here...

https://github.com/projectchrono/chrono/blob/a775004f5e26f16ffe89f0554b58077c551f8734/src/chrono_vehicle/terrain/RigidTerrain.cpp#L493-L499

To me this caused troubles because I had set the collision mask of a certain body to exclude collisions with my vehicle chassis (family 0) and was surprised to see that this also switched off collisions with my terrain. 

Thanks for all your great work you put into Chrono, 
Christian

rserban, 2021-07-14:
The intent was to only disable potential contact forces between different patches.
But I see your point. I pushed a fix.

closed 2021-07-14 by rserban

chfeller, 2021-07-14:
Hi Radu, thanks a lot! Now, while looking at your commit, I finally also saw the line "Used only if defining two or more patches" in the header file. So it was on purpose after all - sorry for missing that in the first place. But I still like your fix :)

commit f3423cc0 (2021-07-14): Set collision family for RigidTerrain patches even if only one patch is defined

Fixes #328