projectchrono/chrono #333: C++11 compatibility - variable template

alhirzel, 2021-09-01:
This line causes an issue in GCC 4.8.5. Perhaps this templating requires >C++11 and should be guarded by `CH_ENFORCE_CXX11`? See this article: [variable template](https://en.cppreference.com/w/cpp/language/variable_template).

https://github.com/projectchrono/chrono/blob/fd8095210d8810069730a3fd766a90af8bad1fda/src/chrono/utils/ChUtilsSamplers.h#L234-L235

rserban, 2021-09-01:
This should not be an issue with a newer version of GCC.
Note that we require GCC 4.9 or newer (see [https://api.projectchrono.org/tutorial_install_chrono_linux.html](https://api.projectchrono.org/tutorial_install_chrono_linux.html)) but the truth is all of us are using much newer versions.

alhirzel, 2021-09-01:
Noted. Is the GCC 4.9 requirement as of a certain version of Chrono?

Note that some widely-deployed distributions are currently using GCC 4.8.5, e.g. CentOS 7/RHEL 7.