projectchrono/chrono #340: Library not loaded in the pychrono/_core.so

domino2, 2021-10-11:
Hello, I did fresh installation on my Mac and I've encountered with the running issue.
My script failed on the first line `import pychrono as chrono` where after run under the conda environment I've got this error message

```
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import pychrono as chrono
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/pychrono/__init__.py", line 44, in <module>
    from .core import *
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/pychrono/core.py", line 13, in <module>
    from . import _core
ImportError: dlopen(/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/pychrono/_core.so, 2): Library not loaded: /usr/local/opt/libomp/lib/libomp.dylib
  Referenced from: /usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/pychrono/_core.so
  Reason: image not found
```

I'm running chrono develop.

domino2, 2021-10-12:
I think it may relates to the platform issue. So I'm going to build the python module and will see if I move the stone.

Benatti1991, 2021-10-15:
@domino2  it might be due to unsupported CPU instructions (typically AVX2) that are used when building Conda packages. Building yourself would solve the issue, please report back if so (or if not so)
Thanks,
Simon