projectchrono/chrono #348: A question about tutorials

dcagritan, 2021-11-17:
Dear all,
I'm quite a newbie here and trying to understand Chrono. First of all, thanks for all your efforts. Chrono seems very promising!
I've just compiled Chrono and now I'd like to run some tutorials. For instance, I'd like to run demo_VEH_HMMWV_DefSoil, however, I don't know how to do that. Here: chrono/build_chrono/src/demos/vehicle/demo_HMMWV/CMakeFiles/demo_VEH_HMMWV_DefSoil.dir
I have this folder, however, I don't know how to proceed. I guess that somehow I have to compile the tutorials too, but I'm kind of lost and would really appreciate your help.
Thank you and Kind Regards

rserban, 2021-11-17:
All demo executable are under a `bin/` subdirectory of your build tree (if using MSVC, under a further subdirectory of `bin/` corresponding to the build configuration: Release, Debug, etc).  Simply navigate to that directory and run the demo from there.

dcagritan, 2021-11-17:
Hello @rserban,
thank you very much for your answer. Now it works. However, I would like to ask you how to create a demo file like this one. For instance, I'd like to modify the soil mesh a bit and perhaps the tires of the vehicle as well. I'd be glad if you could help me on that too.
Thank you and Kind Regards,

rserban, 2021-11-17:
Look at [https://api.projectchrono.org/tutorial_install_project.html](https://api.projectchrono.org/tutorial_install_project.html) which explains hot to use Chrono from your own program.

dcagritan, 2021-11-17:
Thank you @rserban!!
Best regards

closed 2021-11-21 by rserban