projectchrono/chrono #349: Understanding the Deformable FEA vehicle terrain

dcagritan, 2021-11-23:
Dear all,
I'm trying to understand how you derive the deformable FEA in Chrono. I know quite a bit about finite element methods, however, I come from fluid mechanics background and would really appreciate your help. First of all, what are the governing equations here? I mean the equation that we multiply with the test functions to have the weak form and furthermore limit that weak form to finite space(Galerkin Form)? I assume that the constitutive equation is the Drucker-Prager plasticity, am I right on that? You say that FEADeformableTerrain employs ChElementHexaANCF_3813_9 type of element, however, I can't find that element on Chrono installation folder. Do you have any publication that you talk about this finite element formulation applied onto deformable terrain?
Thank you and have a great day!
Regards,
Deniz

rserban, 2021-11-27:
There are different FEA formulations used in Chrono, most notable co-rotational and ANCF.  You can find some details about these in the whitepapers provided with the Chrono documentation: [https://projectchrono.org/whitepapers/](https://projectchrono.org/whitepapers/).
As for the special "brick_9" element used to model deformable soil with FEA, see this [ASME conference paper](https://www.researchgate.net/publication/311463249_Modeling_of_Deformable_Tire_and_Soil_Interaction_Using_Multiplicative_Finite_Plasticity_for_Multibody_Off-Road_Mobility_Simulation).
In the latest Chrono code, ANCF element names were changed for consistency.  You are probably looking at the latest documentation (which uses the new name `ChElementHexaANCF_3813_9`) and at an older code base (where this element was named `ChElementBrick_9`).
Finally, note that we rarely (if at all) use FEA to model deformable soil.  Instead, we use granular dynamics, continuum representations solved using SPH, or semi-empirical methods (such as SCM).

dcagritan, 2021-11-27:
Hello Professor Serban,
thank you very much for your detailed answer! May I further ask you the reason why you don't employ FEM to model deformable soil?
Thank you very much and wish you a nice weekend.
Regards,
Deniz

rserban, 2021-11-28:
Primarily computational cost, difficulty of obtaining parameters, dealing with very large soil deformations and fracture/ejecta.
But it may be a valid approach for certain applications/scenarios.  And interesting hybrid approach is described in a couple of JCND papers: [1](https://www.researchgate.net/publication/330348697_Hierarchical_Multiscale_Modeling_of_Tire-Soil_Interaction_for_Off-Road_Mobility_Simulation) and [2](https://www.researchgate.net/publication/351883008_Enhancing_Hierarchical_Multiscale_Off-Road_Mobility_Model_by_Neural_Network_Surrogate_Model).

dcagritan, 2021-11-29:
Hello Professor,
thank you very much for your detailed answer!! I'd previously searched for more research papers but somehow was unsuccessful. I'll definitely read all these papers in great detail!
Professor, I'd like to ask you a few more questions and I'd be very glad if you could answer me. You say that obtaining parameters is difficult with FEM. However, these are the basic soil properties, right? I don't understand why is it difficult to obtain them. And moreover, if they are difficult for FEM, they should be difficult for DEM, SCM, or any other method too. Am I missing something here? 
Besides that, you say that large soil deformations would be problematic. Is FEM not capable of capturing large soil deformations? 
Professor, I'd be very glad if you could answer my questions. Thank you very much in advance and wish you a nice day.
Best regards,
Deniz