projectchrono/chrono #350: ImportError: undefined symbol

nehelmalhotra, 2021-11-26:
Hello everyone,

Firstly thank you very much for the amazing resource.

I am running into some issues when trying to run the demo files and would really appreciate your help.

`ImportError: /miniconda3/envs/asp/lib/python3.8/site-packages/pychrono/../../../libChronoEngine_cascade.so: undefined symbol: _ZNK13Poly_Triangle3GetERiS0_S0_`

I am using Linux-18.04, python 3.8.

Benatti1991, 2021-11-28:
Hi,
I think you don't have Cascade installed in your Anaconda env. Please, install it using the following command:
`conda install -c dlr-sc opencascade --yes`
And try again. Let me know if it works or not, please.
Thanks,
Simon

nehelmalhotra, 2021-11-28:
> Hi, I think you don't have Cascade installed in your Anaconda env. Please, install it using the following command: `conda install -c dlr-sc opencascade --yes` And try again. Let me know if it works or not, please. Thanks, Simon

Hi Simon, 

Thank you so much for your response. You were indeed correct and that resolved the issue!

closed 2021-11-28 by nehelmalhotra