projectchrono/chrono #354: data folder is installed even when BUILD_DEMOS=OFF

uthidata, 2021-12-26:
What I did:
```
cmake -DBUILD_DEMOS=OFF -DCMAKE_INSTALL_PREFIX="$package_dir" \
-DENABLE_MODULE_COSIMULATION=ON -G Ninja ..
cmake --build . --config Release --target install
```
What I expect:
The libraries (*.so) are built and I could zip the whole folder and use in my build system.

What happened:
The resulting .tar.gz archive is >300MB, because the share/data folder alone is 700MB.
Since demos are not built, I don't think the images and models are needed.

dariomangoni, 2023-12-07:
Hi @uthidata, quite late reply, but to tell you (and anyone else that might be interested) that the _data_ folder is used for example to load fonts, default backgrounds, by the Irrlicht visualization. This happens by default, so it applies to **any** simulation that uses the Irrlicht graphical interface. That's why it's there.

Anyway, you are at the same time right: in that folder there is **much more** than just a couple of fonts and background textures! It's also used by demos, tests, etc... so it's definitely true that it's quite overshooting to copy everything, especially when demos are not requested. Unfortunately these low priorities fixes always get postponed!

closed 2023-12-07 by dariomangoni

uthidata, 2023-12-07:
One question, if I don't enable irrlicht (as per my configuration here), then can I just exclude this folder before packaging? is `data/cosimulation` needed, or is it for tests only?

Besides, if you don't have time, I can dig deeper and fix it and do a PR.

rserban, 2023-12-07:
The data/ directory contains files that are required for other modules, besides Chrono::Irrlicht.  
And of course, most demos use data/ in one form or another.
You can certainly look for a solution to your particular needs (if space is an issue), but I'm not sure a change to Chrono is warranted.

dariomangoni, 2023-12-07:
Data folder is used also for **vehicle** and **models** (I just gave Irrlicht as an example but the thing is a little more complex).
(if you search for "GetChronoData" usages in the source you will get a good idea).
So it's a little trickier and not sure if totally worth now.

Anyway, if you need to shrink the size of your build package (or even better, the install folder) you can surely cherry pick only the files that you actually need (potentially even none!)