projectchrono/chrono #356: Understanding the SCM Soil Displacement

dcagritan, 2022-01-14:
Dear Chrono Team,
first of all, thank you very much for this amazing work!!
Recently I've been trying to understand SCM soil model in general and its implementation in Chrono Engine. I understand that it combines Bekker's empirical models and other theories and calculates the forces transmitted from the soil onto the vehicle. However, to do this, we have to know about the displacement of the soil or in other words as in your technical paper, the sinkage. There I have some issues understanding the working mechanism of the SCM. How do you know about the sinkage without having any knowledge about the vehicle? I see that you do some raycasting to detect where the tyre is and subtract it from the initial soil height to define the sinkage. But how does the tyre stop there in the first place? How do you calculate how much the tyre gets until that height and not more. What defines this? Could you please help me with this?
Thank you and kind regards,
Deniz

rserban, 2022-01-15:
Deniz, 
In a nutshell, the pressure increases with sinkage and so, at some value of sinkage, the terrain forces will balance the weight of the vehicle.  Note that this is the exact same with the original Bekker formula (which has no information about the vehicle, other than the 'b' dimension of the tire).
By the way, this is similar to how contact forces develop in a soft-body (penalty-based) approach; the penetration increases until the resulting normal contact force is enough to stop motion in that direction.  Or how the suspension of a vehicle works: the spring compresses until the resulting elastic force balances the weight of the vehicle (or the resultant of whatever forces act on the vehicle) - no need to know the mass of the vehicle when implementing the suspension element.
--Radu

dcagritan, 2022-01-19:
Dear Professor Serban,
thank you very much for your great explanation!! I understand it now.
I wish you a great day!
Kind Regards,
Deniz

closed 2022-01-19 by rserban