projectchrono/chrono #359: Build fails: fatal error: 'BRepAdaptor_HSurface.hxx' file not found

yurivict, 2022-01-27:
```
/disk-samsung/freebsd-ports/science/chrono/work/chrono-7.0.1/src/chrono_cascade/ChCascadeMeshTools.cpp:37:10: fatal error: 'BRepAdaptor_HSurface.hxx' file not found
#include <BRepAdaptor_HSurface.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

Where is ```BRepAdaptor_HSurface.hxx``` expected to come from?

Version: 7.0.1
OS: FreeBSD 13

rserban, 2022-01-27:
Open Cascade.

Did you provide the proper path to the (CMake) Open Cascade project configuration script during Chrono cmake configuration?
See https://api.projectchrono.org/module_cascade_installation.html.  Note that we currently do not support Open Cascade 7.5.0.
Please post your CMakeCache.txt.

rserban, 2022-01-27:
I should have been more clear.  Currently, we only support Open Cascade 7.4.0.    
7.5.0 and 7.6.0 are not supported yet (precisely because API changes such as this one).

yurivict, 2022-01-27:
Do you have a branch that supports the latest Open Cascade?

The FreeBSD port for Open Cascade has been updated a while ago and now Chrono fails to build.

rserban, 2022-01-27:
No, not yet.

yurivict, 2022-01-27:
Is there an ETA?

rserban, 2022-01-27:
Unfortunately, no.  I don't have the time right now and don't know if anyone else on our side is planning on doing that anytime soon.  Having said that, we'd be more than happy to accept a pull request with this update (as long as the code also supports the current 7.4.0 version).

yurivict, 2022-01-28:
FYI: OpenCascade-7.4.0 doesn't even build with current cmake. I tried to resurrect the old version and found that it fails to build now.

rserban, 2022-01-29:
Can you please provide more details?  It works fine on the systems we are testing (Ubuntu, arch, Windows) but I do not have access to a FreeBSD system.
Note also that the cmake mechanism for finding and configuring Open Cascade in Chrono has changed.  You must only provide the path to the directory which contains the Open Cascade configuration script (OpenCASCADEConfig.cmake) through the CMake variable `OpenCASCADE_DIR`.  On my Ubuntu installation, this is `/usr/local/lib/cmake/opencascade`.

yurivict, 2022-01-29:
Ubuntu doesn't update packages as often. Everything is older on Ubuntu.

cmake in particular was updated recently, and it fails to configure older Open Cascade:
```
===>  Performing out-of-source build
/bin/mkdir -p /wrkdirs/usr/ports/cad/opencascade740/work/.build
-- The C compiler identification is Clang 12.0.1
-- The CXX compiler identification is Clang 12.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info. Detecting doxygen
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) (Required is at least version "1.8.4")
-- Info: TCL is used by OCCT
-- Found Tclsh: /usr/local/bin/tclsh8.6 (found version "8.6")
-- Info: TK is used by OCCT
-- Info: Freetype is used by OCCT
-- Found Freetype: /usr/local/lib/libfreetype.so (found version "2.11.0")
CMake Error at adm/cmake/vtk.cmake:159 (endif):
  Flow control statements are not properly nested.
Call Stack (most recent call first):
  CMakeLists.txt:14 (include)
  CMakeLists.txt:523 (OCCT_INCLUDE_CMAKE_FILE)

-- Configuring incomplete, errors occurred!
```

yurivict, 2022-01-30:
I managed to resurrect OpenCascade-7.4.0 (thanks for the hint with VTK).

But please add support OpenCascade >7.4.0 when this would be possible.

thepianoboy, 2022-04-17:
This seems to be related to #357.

closed 2025-11-26 by uebian