projectchrono/chrono #362: Support macOS on Apple M1 hardware

thepianoboy, 2022-02-03:
Some users have reported issues building Chrono on macOS on Apple silicon. `-march=native` isn't supported by clang on non-x86 platforms, so that needs to be remedied.

See the [mailing list](https://groups.google.com/g/projectchrono/c/xuy3VyiWyYE) for the original post.

---

I suspect this can be fixed by changing the default behavior in CMake to `-mtune=native` instead, but I don't have a recent mac to test it on. Hopefully one of our mac devs has recent enough hardware to test on.

thepianoboy, 2022-02-03:
I tagged our resident Mac users @rainergericke and @AaronYoung5 just in case they can provide some more insight.

rainergericke, 2022-02-03:
Chrono works well on arm64 Macs. What doesn't work is SIMD-support. Mac users should use this setting in cmake:

cmake -G "Xcode" -DUSE_SIMD:BOOL=OFF ....

Maybe we find a way to set it automatically...

thepianoboy, 2022-02-03:
> Chrono works well on arm64 Macs. What doesn't work is SIMD-support.

When I ported chrono to arm64 Linux, I was able to get simd working. If I had to guess, we're missing something Mac-specific like an apple clang flag or an Xcode setting.

> Maybe we find a way to set it automatically...

I know how to detect that in CMake, I'll push a (temporary?) fix later today.

rainergericke, 2022-02-04:
Thanks to Enrique's proposal, I cout find a possible solution. The wrong compiler flags are set by some find*.cmake files.
Cmake now idendifies the NEON instruction set for the M1 processor and building runs through. I saw a small performance benefit of about 4% compared to the configuration with SIMD switched off (70 unit tests run by 'ctest').

With my attached files we brake the build on Intel Macs yet. So we need an cmake expert, who modifies the configuration files so that M1 as well as Intel get the correct compiler settings
[Archiv.zip](https://github.com/projectchrono/chrono/files/8002915/Archiv.zip)
.

closed 2022-02-09 by rserban

yhppark902, 2022-02-09:
Thanks for the quick issue support. But, I got a mesage with matlab.
ld: warning: ignoring file /Applications/MATLAB_R2021b.app/extern/../bin/maci64/libeng.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /Applications/MATLAB_R2021b.app/extern/../bin/maci64/libmat.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /Applications/MATLAB_R2021b.app/extern/../bin/maci64/libmx.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_engClose", referenced from:
      chrono::ChMatlabEngine::~ChMatlabEngine() in ChMatlabEngine.cpp.o
  "_engEvalString", referenced from:
      chrono::ChMatlabEngine::Eval(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_engGetVariable", referenced from:
      chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_engOpen", referenced from:
      chrono::ChMatlabEngine::ChMatlabEngine() in ChMatlabEngine.cpp.o
  "_engPutVariable", referenced from:
      chrono::ChMatlabEngine::PutVariable(Eigen::Ref<Eigen::Matrix<double, -1, -1, 1, -1, -1> const, 0, Eigen::OuterStride<-1> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_engSetVisible", referenced from:
      chrono::ChMatlabEngine::SetVisible(bool) in ChMatlabEngine.cpp.o
  "_mxCreateDoubleMatrix_800", referenced from:
      chrono::ChMatlabEngine::PutVariable(Eigen::Ref<Eigen::Matrix<double, -1, -1, 1, -1, -1> const, 0, Eigen::OuterStride<-1> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_mxDestroyArray_800", referenced from:
      chrono::ChMatlabEngine::PutVariable(Eigen::Ref<Eigen::Matrix<double, -1, -1, 1, -1, -1> const, 0, Eigen::OuterStride<-1> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
      chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_mxGetDimensions_800", referenced from:
      chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_mxGetNumberOfDimensions_800", referenced from:
      chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
  "_mxGetPr_800", referenced from:
      chrono::ChMatlabEngine::PutVariable(Eigen::Ref<Eigen::Matrix<double, -1, -1, 1, -1, -1> const, 0, Eigen::OuterStride<-1> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
      chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in ChMatlabEngine.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libChronoEngine_matlab.dylib] Error 1
make[1]: *** [src/chrono_matlab/CMakeFiles/ChronoEngine_matlab.dir/all] Error 2
make: *** [all] Error 2

rainergericke, 2022-02-09:
> chrono::ChMatlabEngine::Eval(std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator >) in ChMatlabEngine.cpp.o
> chrono::ChMatlabEngine::GetVariable(Eigen::Matrix<double, -1, -1, 1, -1, -1>&, std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator >) in ChMatlabEngine.cpp.o
> chrono::ChMatlabEngine::PutVariable(Eigen::Ref<Eigen::Matrix<double, -1, -1, 1, -1, -1> const, 0, Eigen::OuterStride<-1> > const&, std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator >) in ChMatlabEngine.cpp.o

Hi,

MATLAB is actually a x86_64 application. So you connot link parts of it to arm64 code. Mathworks plans to change to arm64. We have to wait for it.

Best

Rainer Gericke

commit 0da8189d (2022-02-09): Fix for NEON support on Apple M1 processors

Fixes #362