projectchrono/chrono #363: Do all modules of Chrono Engine work on Parallel?

dcagritan, 2022-02-04:
Dear Chrono Team,
I'm asking this question because the demo file demo_VEH_HMMWV_DefSoil, and it has the function call system->SetNumThreads(std::min(8, ChOMP::GetNumProcs()));, and I was assuming that it was already working in parallel. However, I've realized that Chrono has another module called PARALLEL. I'm a bit confused. Do I have to employ PARALLEL to have multi-thread support or Chrono already works partially in parallel without using PARALLEL module. 
Thank you and kind regards,
Deniz

rserban, 2022-02-05:
Deniz, 

Could you please post this (and any future similar) questions to the Chrono user mailing list.?

While some other projects use the GitHub issues for this kind of support, we prefer to reserve this for bug reports and feature requests -- that way we can close them when needed.   

Once closed, a GitHub issue is less likely to be found by other interested users, while posts to the Google Group are a better place for this type of questions.

I will provide an answer as soon as you post this on the mailing list.

Thanks.

dcagritan, 2022-02-05:
Hello Professor Serban,
sure! I just did that.
I'm closing this issue now. Thank you very much and wish you a nice day!
Regards,
Deniz

closed 2022-02-05 by dcagritan