projectchrono/chrono #365: Unable to run Pychrono on M1 mac

ishaanamahajan, 2022-02-14:
I tried importing Pychrono using the command - `python -c 'import pychrono'`, but got an error saying - `zsh: illegal hardware instruction  python -c 'import pychrono'`.

I am using the Macbook Air M1 (2020).

AaronYoung5, 2022-02-14:
@ishaanamahajan what version of python? And what conda package were you using (you can run `conda list | grep pychrono` to check)?

ishaanamahajan, 2022-02-14:
> @ishaanamahajan what version of python? And what conda package were you using (you can run `conda list | grep pychrono` to check)?

pychrono                  6.0.0                  py38_638    projectchrono/label/develop

thepianoboy, 2022-02-14:
While Chrono currently _does_ work on M1 Macs, PyChrono packages are not built for that target. Until such distributions are available, you will need to build Chrono's Python module for yourself.

I have some ideas about how we might be able to support packaging for M1, but they'll take some time to test (on the order of days/weeks). I'll follow up with what I find out.

uebian, 2025-10-25:
Thank you for raising the issue. Our conda package building pipeline has recently changed and we haven't updated the document. We have simplified the installation process so you won't need to manually download and install packages as before. Could you please try using the following two command in the NEW environment to install PyChrono through conda? Thank you very much!

```
conda create -n chrono_test python=3.12 -c conda-forge
conda install projectchrono:pychrono -c dlr-sc -c conda-forge
```

Let me know if you still encounter issues.

closed 2025-11-26 by uebian