projectchrono/chrono #370: a question about tire force

liufuda, 2022-03-10:
dear chrono team :
 i run demo_VEH_Sedan whith full throttle ,then export  the forward left tire force at x, why the force discontinuity？
expect your reply.thanks!
[image]

liufanfanliu, 2022-08-19:
It depends what kind of tire model you use, the pac02 tire has bad performance in vehicle simulation and need to be improve. The tmeasy tire does not has discontinuity

hanhantuozi, 2023-01-01:
> dear chrono team : i run demo_VEH_Sedan whith full throttle ,then export the forward left tire force at x, why the force discontinuity？ expect your reply.thanks! [image]

Could you teach me how to export the force? :)

rserban, 2023-01-04:
@hanhantuozi  - Simply invoke the functions [ChTire::ReportTireForce,](https://api.projectchrono.org/classchrono_1_1vehicle_1_1_ch_tire.html#adad1ae919475e9a394536c7117eea40a) passing a pointer to the terrain object.  See for example [demo_VEH_HMMWV_Accel](https://github.com/projectchrono/chrono/blob/b20c43f5936d897b29af32f17fdf1d98262f0628/src/demos/vehicle/wheeled_models/demo_VEH_HMMWV_Accel.cpp#L207) for usage in an actual demo.

closed 2023-01-04 by rserban