projectchrono/chrono #371: Fails on arm64 architecture: error: the clang compiler does not support '-march=native'

yurivict, 2022-03-20:
Build of 7.0.1 fails on arm64:
```
c++: error: the clang compiler does not support '-march=native'
```

clang-12
OS: FreeBSD 13

thepianoboy, 2022-03-23:
This is related to #362. It looks like the fix that we made was macOS-specific and doesn't work for Clang other platforms. 

I will see if I can get an arm64 FreeBSD image running on one of the testing platforms and try to produce a generalized solution. It might take a while if I need to use an emulated CPU, though.

thepianoboy, 2022-04-13:
I finally got my hands on an aarch64 system. :tada: 

I'm having trouble running FreeBSD on this particular board, but that might not be an issue... the Linux image I have on hand returns this:
```
clang-13: error: the clang compiler does not support '-march=native'
```

From what I can tell, the bug still affects pretty much anything running Clang on 64-bit Arm that isn't a Mac. It seems to be the same error for both Linux and FreeBSD so the fix just needs to be generalized for more than one OS. Looking at my build output, I see that the offending flag is set more than once; I will probably need to resolve that as well.

thepianoboy, 2022-04-13:
It's also worth noting that the bug affects both release/7.0 and develop.

thepianoboy, 2022-04-17:
Fixed in release 7.0.3

closed 2022-04-17 by thepianoboy