projectchrono/chrono #375: make install error

IvanVnucec, 2022-04-05:
After successfull build, I've ran `make install` but I've got an error saying:
```sh
[ 98%] Built target demo_IRR_rev_trans
[ 99%] Built target demo_POST_emitter_asset
[ 99%] Built target demo_POST_emit_creation
[100%] Built target demo_POST_timestepping
[100%] Built target demo_POST_gnuplot
[100%] Built target demo_POST_povray
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/share/chrono/data
CMake Error at cmake_install.cmake:46 (file):
  file INSTALL cannot make directory "/usr/local/share/chrono/data": No such
  file or directory.

make: *** [Makefile:127: install] Error 1
```

IvanVnucec, 2022-04-05:
I've solved the issue by running `make install` with sudo privileges.
It should be somewhere written that you need to install it with sudo privileges. I know that its standard procedure, but still...

rserban, 2022-04-05:
Well, that depends on **where** you want to install Chrono (or any other package for that matter).
You need root privileges **only** if you want to install in a system directory (such as `/usr/local/share` in your case).
But you could choose to install in a location where you have write permissions as a user.

closed 2022-04-05 by rserban

IvanVnucec, 2022-04-05:
sure but that should be written somewhere in the install documentation, no?

rserban, 2022-04-05:
This is not a Chrono issue.  You are running `make` at that point, so maybe that should be specified in the documentation of your make program (although, I suspect that even the documentation of something like GNUMake doesn't explicitly mention that, for the exact same reason I mentioned before).