projectchrono/chrono #377: cannot build template_project on Ubuntu 21.10

IvanVnucec, 2022-04-06:
I've built and succesfully isntalled Chrono as specified in the Install documentation.
I've copied template_project and cd into it,
made build foder, cd into it, 
ran `cmake -DChrono_DIR=<path_to_cloned_chrono_dir>/cmake ..`,
I get missing Chrono_DIR error.
Here is the full cmake setup console output:
```sh
$ cmake -DChrono_DIR=<path_to_cloned_chrono_dir>/cmake ..
-- Could NOT find Chrono (missing: Chrono_DIR)
Could not find Chrono or one of its required modules
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ivan/Desktop/template_project/build
```
I've also tried with CMake GUI by running `ccmake ..` in the build foder. I've set the Chrono_DIR variable to <path_to_cloned_chrono_dir>/cmake but still I get missing: Chrono_DIR error.

IvanVnucec, 2022-04-06:
I've managed to compile the template_project by running `ccmake ..` in the build folder and with configuration as follows:
```
 CMAKE_BUILD_TYPE                 Debug
 CMAKE_INSTALL_PREFIX             /usr/local
 Chrono_DIR                       /usr/local/lib/cmake
```
Then after pressing c to configure and then g to generate, I've ran `make` and got this successfull build message:
```sh
$ make
Scanning dependencies of target myexe
[ 50%] Building CXX object CMakeFiles/myexe.dir/my_example.cpp.o
[100%] Linking CXX executable myexe
[100%] Built target myexe
```

IvanVnucec, 2022-04-06:
After successfull build I've ran `myexe` compiled program and got the following runtime error:
```sh
$ ./myexe
./myexe: error while loading shared libraries: libChronoEngine_postprocess.so: cannot open shared object file: No such file or directory
```

How do I solve it?

IvanVnucec, 2022-04-06:
I have Postprocess module already installed because I've followed https://api.projectchrono.org/tutorial_install_chrono_linux.html where it says that the ENABLE_MODULE_POSTPROCESS should be set to ON.

IvanVnucec, 2022-04-06:
I've managed to solve the issue by running `sudo ldconfig` as recommended [here](https://stackoverflow.com/a/28460293/13789930).

thepianoboy, 2022-05-20:
> Maybe the `ldconfig` command should be added to the chrono install documentation as standard procedure so someone would not loose its precious time figuring out what the problem might be.

Running `ldconfig` for chrono may be a distribution-specific fix, but it's worth noting. My system picks it up automatically, but that might be due to some customization on my end.