projectchrono/chrono #383: Issue with installing pychrono

harshamarupudi56, 2022-04-14:
Hello I am currently attempting to create a conda environment to run a repository that makes use of pychrono as a dependency. However I have been receiving this error regarding pychrono. lib64limb.so.6 GLIBC 2.66. I am currently on Python 3.8. I have read that a solution is to revert Python to version 3.7, however I have seen users who are able to utilize it on Python 3.8 so I do not think that is the issue.

thepianoboy, 2022-05-20:
Could you please let us know what OS (including distribution and version, where applicable) you are running on, what conda channels you are using, as well as any non-default packages you have in your environment?

qiancao, 2022-06-03:
Hi @thepianoboy, thank you for responding. I work with @harshamarupudi56 and can respond to some of these questions. We're working on a computer with Ubuntu 18.04 installed.

The packages are installed in a fresh conda environment with: 

`conda env create -f environment.yml`

where the yml file is defined as:

```
name: chronoenv
channels:
  - conda-forge
dependencies:
  - python
  - numpy
  - scipy
  - numba
  - mkl
  - matplotlib
  - scikit-image
  - scikit-learn
  - spyder
  - ipython
  - glob2
  - projectchrono::pychrono
  - pip
  - pip:
    - pynrrd # image/data file formats
    - pydicom
    - h5py
    - nibabel
    - meshio
    - trimesh
    - tetgen
    - vtk
    - pyvista
    - pyvistaqt
    - pyradiomics
    - umap-learn 
    - edt 
    - skan
    - cvxopt # convex optimization
```
  
When trying to import pychrono, the error message reads:

`File "FEA/fea.py", line 22, in <module>
import pychrono as chrono
File "/home/usr/anaconda3/envs/chronoenv/lib/python3.8/site-packages/pychrono/__init__.py", line 44, in <module>
from .core import *
File "/home/usr/anaconda3/envs/chronoenv/lib/python3.8/site-packages/pychrono/core.py", line 13, in <module>
from . import _core
ImportError: /lib64/libm.so.6: version `GLIBC_2.29' not found (required by /home/usr/anaconda3/envs/chronoenv/lib/python3.8/site-packages/pychrono/_core.so)`

The library complains GLIBC 2.29 not found because Ubuntu 18.04 does not use GLIBC 2.29. From what I've read, it is also not possible to upgrade GLIBC without breaking the OS. Could we patch this such that the requirement on GLIBC is not so strict? I think Ubuntu 18.04 comes with GLIBC 2.17.

Thanks!

uebian, 2025-11-08:
Thank you for raising the issue. Our conda package building pipeline has recently changed and we haven't updated the document. We have simplified the installation process so you won't need to manually download and install packages as before. Could you please try using the following two command in the NEW environment to install PyChrono through conda? Thank you very much!

```
conda create -n chrono_test python=3.12 -c conda-forge
conda install pychrono -c projectchrono -c dlr-sc -c conda-forge
```