projectchrono/chrono #388: How did you achieve lidar motion blur

TigerJieZ, 2022-04-26:
Hi, I read on the Lidar introduction page that you use keyframing and motion simulation to take into account the time difference in fiber scanning. Could you please introduce how this part is designed in more detail? Or can you tell me where the relevant code is? I looked at your code :ChLidarSensor. CPP has no meaningful code.

TigerJieZ, 2022-04-26:
thank you for your reply. I have not successfully used the Sensor function, so there is a basic question: Is the Sensor in Projectchrono rendered in real time? In other words, will the real time of 10 minutes rendering the sensor data in 10min?

rserban, 2022-04-27:
@TigerJieZ  - I am sure Asher will follow up with answers to your questions.  However, I would like to suggest you use our users mailing list for this kind of discussions.  We have not set up the GitHub issues for this type of questions.

The Chrono user mailing list is at: https://groups.google.com/g/projectchrono

TigerJieZ, 2022-04-27:
> @TigerJieZ - I am sure Asher will follow up with answers to your questions. However, I would like to suggest you use our users mailing list for this kind of discussions. We have not set up the [...]

got, thanks

amelmquist, 2022-04-27:
> thank you for your reply. I have not successfully used the Sensor function, so there is a basic question: Is the Sensor in Projectchrono rendered in real time? In other words, will the real time of 10 minutes rendering the sensor data in 10min?

This entirely depends on the problem you are trying to setup and the hardware you are using to solve it. For a camera, it depends on size of the image, scene complexity, lights in the scene, and GPU. Its hard to give a more specific answer as we don't try to maintain a specific frame rate. For example, in a benchmark on a GTX 1080, we run 729 instanced meshes for a total of ~125M triangles at 1080p 60Hz and takes about 30 seconds to simulation 10s. That shouldn't be extrapolated, but should give some idea.

> After reading lidar.cu and choptixgeometry.cpp, I figured out that Chrono simulated motion distortion by setting different launch time for each rays. However, I still don't quite understand what the "key frame" refers to. Can you describe more detailed information?

For keyframes and motion transforms, I suggest looking at the OptiX guide here: [https://raytracing-docs.nvidia.com/optix7/guide/index.html#acceleration_structures#motion-blur](https://raytracing-docs.nvidia.com/optix7/guide/index.html#acceleration_structures#motion-blur)

If you have followup questions, please post to the forum and we can continue the discussion. Thanks!

closed 2022-04-27 by amelmquist