projectchrono/chrono #389: Build fails on arm64: /usr/local/bin/ld: cannot find -lFALSE

yurivict, 2022-05-08:
At least versions 7.0.3, 7.0.2, 7.0.1 fail:
```
[ 44%] Linking CXX shared library ../../lib/libChronoEngine_mumps.so
cd /wrkdirs/usr/ports/science/chrono/work/.build/src/chrono_mumps && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/ChronoEngine_mumps.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include -pthread -fopenmp=libomp -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include -pthread -fopenmp=libomp  -march=armv8-a -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include -pthread -fopenmp=libomp  -march=armv8-a -DNDEBUG  -Wl,-rpath=/usr/local/lib/gcc10  -L/usr/local/lib/gcc10 -B/usr/local/bin -fstack-protector-strong -shared -Wl,-soname,libChronoEngine_mumps.so -o ../../lib/libChronoEngine_mumps.so CMakeFiles/ChronoEngine_mumps.dir/ChSolverMumps.cpp.o CMakeFiles/ChronoEngine_mumps.dir/ChMumpsEngine.cpp.o   -L/opt/intel/compilers_and_libraries/linux/mkl/lib/intel64  -L/opt/intel/compilers_and_libraries/linux/compiler/lib/int
 el64  -L/wrkdirs/usr/ports/science/chrono/work/intel/compilers_and_libraries/linux/mkl/lib/intel64  -L/wrkdirs/usr/ports/science/chrono/work/intel/compilers_and_libraries/linux/compiler/lib/intel64  -Wl,-rpath,/opt/intel/compilers_and_libraries/linux/mkl/lib/intel64:/opt/intel/compilers_and_libraries/linux/compiler/lib/intel64:/wrkdirs/usr/ports/science/chrono/work/intel/compilers_and_libraries/linux/mkl/lib/intel64:/wrkdirs/usr/ports/science/chrono/work/intel/compilers_and_libraries/linux/compiler/lib/intel64:/wrkdirs/usr/ports/science/chrono/work/.build/lib:/usr/local/lib: ../../lib/libChronoEngine.so /usr/local/lib/libmumps_common.so /usr/local/lib/libdmumps.so -lFALSE /usr/lib/libomp.so -lpthread -lpthread 
/usr/local/bin/ld: cannot find -lFALSE
c++: error: linker command failed with exit code 1 (use -v to see invocation)
*** [lib/libChronoEngine_mumps.so] Error code 1
```

clang-13
OS: FreeBSD 13

thepianoboy, 2022-05-20:
I do not have access to FreeBSD on arm64, but I'd like to rule out whether this is a general issue. Could you please share your CMakeCache.txt (feel free to sanitize any personal information) and any relevant packages you have installed so I can attempt to reproduce the error?

yurivict, 2022-05-20:
> Could you please share your CMakeCache.txt [...]

Since the failure occurs on remote build server with ARM64 architecture I don't have access to CMakeCache.txt.

It succeeds on amd64.

thepianoboy, 2022-05-20:
Is there by chance a build script you could share? Anything I can consult to reproduce the build environment, really.

> It succeeds on amd64.

That's helpful information. I _do_ have systems running arm64 macOS and arm64 Linux that I can test on, so I will limit my attempts to those.

yurivict, 2022-05-20:
There is no build script.
The failure occurs when the FreeBSD port is built on arm64.
The ports framework supplies cmake arguments and environment.