projectchrono/chrono #391: Build breaks with flatbuffers-2.0.6: error: no matching member function for call to 'VerifyField'

yurivict, 2022-05-14:
```
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/agent/SynAgent.h:27:
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynMessage.h:24:
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynMessageUtils.h:24:
/wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynFlatBuffers_generated.h:461:12: error: no matching member function for call to 'VerifyField'
--- src/chrono_synchrono/CMakeFiles/ChronoEngine_synchrono.dir/controller/SynControllerFunctions.cpp.o ---
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynSPATMessage.h:24:
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynMessage.h:24:
In file included from /wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynMessageUtils.h:24:
/wrkdirs/usr/ports/science/chrono/work/chrono-7.0.3/src/chrono_synchrono/flatbuffer/message/SynFlatBuffers_generated.h:461:12: error: no matching member function for call to 'VerifyField'
           VerifyField<double>(verifier, VT_TIME) &&
           ^~~~~~~~~~~~~~~~~~~
```

You need to keep the flatbuffers format definition in the repository and regenerate headers during build.

See this  [comment](https://github.com/google/flatbuffers/issues/7272#issuecomment-1124007622) as reference.

ljluestc, 2024-01-05:
```

# Find FlatBuffers package (if not found, install it or specify its path)
find_package(FlatBuffers REQUIRED)

# Specify the path to your .fbs file
set(FBS_FILE "path/to/your/flatbuffers_definition.fbs")

# Generate FlatBuffers C++ headers during build
add_custom_command(
  OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/YourGeneratedFlatBuffers.h
  COMMAND ${FLATBUFFERS_COMPILER} -c -o ${CMAKE_CURRENT_BINARY_DIR} ${FBS_FILE}
  DEPENDS ${FBS_FILE}
)

# Add the generated headers to your project's include directories
include_directories(${CMAKE_CURRENT_BINARY_DIR})

# Add your Chrono target and link it with FlatBuffers
add_executable(YourChronoApp YourSourceFiles.cpp ${CMAKE_CURRENT_BINARY_DIR}/YourGeneratedFlatBuffers.h)
target_link_libraries(YourChronoApp FlatBuffers::FlatBuffers)
```

yurivict, 2024-11-10:
9.0.1 still has the same problem.

This means that the SYNCHRONO module is broken.