projectchrono/chrono #396: Unable to build Synchrono module

mayureshb7, 2022-06-03:
As instructed I have installed OpenMPI that is required to build Synchrono module
When tried to build, it gives me error that MPI is not found (Refer image chrono3)
Can someone help me out with a solution
Please refer attached images for the issue
[image]
[image]
[image]

rserban, 2022-06-03:
Did you by any chance only install the `openmpi-bin` package?  If so, you also need `libopenmpi-dev`.

mayureshb7, 2022-06-04:
> Did you by any chance only install the `openmpi-bin` package? If so, you also need `libopenmpi-dev`.

Thank you for your response, I have installed both 

[image]
[image]

rserban, 2022-06-05:
Could you please check where the OpenMPI headers and libraries are installed (I would expect  them in directories under /usr/lib/x86_64-linux-gnu/openmpi/)?  Are mpicxx and mpiexec in your path?

Please also attach here your CMakeCache.txt file, as well as the full output from cmake when configuring Chrono.

mayureshb7, 2022-06-05:
[CMakeCache.txt](https://github.com/projectchrono/chrono/files/8840019/CMakeCache.txt)
[image]
[image]

[image]
[image]
[image]
[image]

Thank you for your response, 

Please find attahced CMakeCache.txt file and cmake output while configuring Chrono

mpicxx.h is in the path (Refer image) but I was not able to find mpiexec in the path /usr/lib/x86_64-linux-gnu/openmpi/
However, mpiexec was found at another location (Refer images above)

rserban, 2022-06-05:
Odd.  CMake appears to find MPI and yet MPI_FOUND must be set to false...
The only thing I can suggest is trying a newer version of CMake.  I use 3.22 on Ubuntu without issues.

mayureshb7, 2022-06-07:
I uninstalled previous version of Cmake 3.16 and installed it to the latest 3.23.2 still I'm getting the same error.
[image]

rserban, 2022-06-08:
Could you please:
1. Make sure `mpicc` and `mpiexec` are in your path
2. Post here your CMakeError.log and CMakeOutput.log files

closed 2025-11-26 by rserban