projectchrono/chrono #397: Pychrono on Linux

Autonomousanz, 2022-06-04:
Hello
I am unable to run project chrono on linux, I have tried all of the following ;
1. Using the basic installation step as described : https://api.projectchrono.org/tutorial_install_chrono.html
Issue : There is nothing in the linux folder which is the path to be given to irrlicht library 
2. Tried separately installing using sudo apt install libirrlicht1.8 libirrlicht-dev libirrlicht-doc 
3. installed anaconda on linux and then pychrono using https://api.projectchrono.org/development/pychrono_installation.html basic installation steps

I am still facing irrlicht not found and no module named irrlicht error  !

I have done pychrono installation on windows anaconda and it works fine, When I compared the anaconda site packages folder consisting the main pychrono library I found many differences in this folder. Firstly there are some files missing altogether such  as mainly irrlicht.py in pychrono

mayureshb7, 2022-06-04:
Once you install Irrlicht library seperately by sudo apt-get install libirrlicht1.8 libirrlicht-dev libirrlicht-doc and try to do smake in build folder the path automatically is defined for irrlict library, In windows it is different and you have to define it

Autonomousanz, 2022-08-06:
There are some dependency issues with Pychrono on Anaconda with python packages. Having installed different version of python in my environment ,as I found on comparison with working windows anaconda module, there were some files missing in my python site-packages in anaconda on linux.

closed 2022-08-06 by Autonomousanz