projectchrono/chrono #413: POV RAY demo code issues

Autonomousanz, 2022-07-16:
The render_frames.pov generated after running the POV-ray demo file from the project chrono build has many issues to name a few:

    On running the demo_POST_povray from build chrono has Segmentation fault (core dumped) issue when running on ubuntu 20.4 with POV-Ray 3.7.0.8(g++ @ x86_64-pc-linux-gnu).
    The same demo running in ubuntu20.4 based docker image generated output files of stateXXXXX.pov files however while running the render_frames.pov, it is looking for bluewhite. file instead of bluewhite.png in textures folder
[image]
After creating a bluewhite. dummy file in textures, that particular error was resolved just to throw a different error related to code
[image]

starwar9527, 2022-09-08:
I am facing the same issue

rserban, 2022-12-19:
I pushed a fix to this issue.  It is available in the main branch of the repository.  Conda packages including this fix will be uploaded to anaconda.org in a few hours

closed 2022-12-19 by rserban