projectchrono/chrono #422: Cannot open file #include "chrono_thirdparty/chpf/particle_writer.hpp"

simzc, 2022-10-12:
Error C1083 Cannot open include file : “chrono_thirdparty/chpf/particle_writer.hpp”: No such file or directory

rserban, 2022-10-12:
The `CHPF` package is provided as a git submodule.
You need to initialize and update submodules after you clone the chrono repository.

simzc, 2022-10-12:
The problem was solved depend on your comment.  Thanks !

closed 2022-10-12 by rserban

vickmnit, 2022-11-04:
we are facing following error while updating the submodule:

submodule update --force
Cloning into '/mnt/c/Users/VCHOURAS/Downloads/OtherWork/PhysicsEngine/ProjectChrono/chrono/src/chrono_thirdparty/chpf'...
fatal: unable to access '[https://gitlab.com/uwsbel/chpf.git/':](https://gitlab.com/uwsbel/chpf.git/%27:) server certificate verification failed. CAfile: none CRLfile: none
fatal: clone of '[https://gitlab.com/uwsbel/chpf.git'](https://gitlab.com/uwsbel/chpf.git%27) into submodule path '/mnt/c/Users/VCHOURAS/Downloads/OtherWork/PhysicsEngine/ProjectChrono/chrono/src/chrono_thirdparty/chpf' failed
Failed to clone 'src/chrono_thirdparty/chpf'. Retry scheduled

not working even with --no-check-certificate