projectchrono/chrono #423: cannot import name '_core'

1584319, 2022-10-22:
Hello, I did fresh installation on my Windows and I've encountered with the running issue.
My script failed on the  line import pychrono as chrono where after run under the spyder I've got this error message

``````````````````````````````````````````````````````````````````````````````````````````````
runfile('C:/Users/.spyder-py3/temp.py', wdir='C:/Users/LEGION/.spyder-py3')
Traceback (most recent call last):

  File "D:\Anaconda3\envs\chrono\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\legion\.spyder-py3\temp.py", line 4, in <module>
    import pychrono as chrono

  File "D:\Anaconda3\envs\chrono\lib\site-packages\pychrono\__init__.py", line 48, in <module>
    from .core import *

  File "D:\Anaconda3\envs\chrono\lib\site-packages\pychrono\core.py", line 13, in <module>
    from . import _core

ImportError: cannot import name '_core' from partially initialized module 'pychrono' (most likely due to a circular import) (D:\Anaconda3\envs\chrono\lib\site-packages\pychrono\__init__.py)

```````````````````````````````````````````````````````````````````````````````````````````````

Kaiwen-Hong, 2022-12-03:
I also encounter this problem in my Ubuntu 18.04 system.

CosteaPaul, 2023-01-03:
Same:

File "/root/miniconda3/envs/chrono/lib/python3.10/site-packages/pychrono/__init__.py", line 54, in <module>
    from .core import *
  File "/root/miniconda3/envs/chrono/lib/python3.10/site-packages/pychrono/core.py", line 13, in <module>
    from . import _core
ImportError: cannot import name '_core' from partially initialized module 'pychrono' (most likely due to a circular import) (/root/miniconda3/envs/chrono/lib/python3.10/site-packages/pychrono/__init__.py)

Did the conda install with the latest release here:
https://anaconda.org/projectchrono/pychrono/7.0.0/download/linux-64/pychrono-7.0.0-py310_2455.tar.bz2

jeremybk21, 2023-01-21:
I also encounter this issue on windows with python 3.9 & both pychrono v7 and 8. Any solution would be greatly appreciated!

stevewells20, 2023-02-23:
Same here, Ubuntu 20.04, building from source. I've tried on a venv of 3.10 and 3.11, on the OS installation of 3.8. All receive the same error as CosteaPaul.

dariomangoni, 2023-06-21:
Unfortunately we have seen many users affected by this issue (during hands-on sessions it was really a nightmare!) but so far we failed in spotting the cause: apparently some users were affected while others did not. And this was independent from the OS, the Python or the PyChrono version.

While we investigate the issue we suggest those keenest users to build PyChrono by their own from C++ source (section B [here](https://api.projectchrono.org/development/pychrono_installation.html)).
The issue does not show up in this case.

RichiGit97, 2023-06-21:
I had the same problem with the pychrono module on Windows 11, using Python 3.9 and the latest version of anaconda. It seems that the problem has now been resolved by changing the environment variables and adding the anaconda path.
Afterward, open "Anaconda Powershell Prompt" and run `conda init powershell`. Then close anaconda powershell and reopen it.
By doing this, the problem seems to be solved.
I tried running a few demos and they all worked.
[image]

dariomangoni, 2023-06-22:
Thanks Riccardo, it would be great if also other users can confirm that the fix applies also to them!

glennra, 2023-07-13:
I have a similar issue. This is on Fedora37 using chrono I've built from source.

```
(venv) glenn@fedora:~/devel/chrono/build$ env | grep PYTHONPATH
PYTHONPATH=/home/glenn/opt/share/chrono/python
(venv) glenn@fedora:~/devel/chrono/build$ python
Python 3.11.4 (main, Jun  7 2023, 00:00:00) [GCC 12.3.1 20230508 (Red Hat 12.3.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pychrono
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/glenn/opt/share/chrono/python/pychrono/__init__.py", line 63, in <module>
    from .core import *
  File "/home/glenn/opt/share/chrono/python/pychrono/core.py", line 13, in <module>
    from . import _core
ImportError: cannot import name '_core' from partially initialized module 'pychrono' (most likely due to a circular import) (/home/glenn/opt/share/chrono/python/pychrono/__init__.py)
```

glennra, 2023-07-13:
I've  got it working by setting:
`export LD_LIBRARY_PATH=~/opt/lib`

I installed it to `$HOME/opt`.

TroyZhangFawang, 2023-11-11:
"ImportError: cannot import name '_core' from partially initialized module 'pychrono' (most likely due to a circular import)"
Has anyone solved this error? I tried different python version on ubuntu 18.04, but it all failed.

TheCrazyT, 2024-08-29:
Talking about linux systems:
Problem source seems to be an old system with old glibc libraries. (atleast thats what the "ldd"-command on the "_core.so" tells me)
So you only have 2 solutions:
- upgrade the system
- downgrade the pychrono version (and code that uses it)

Not so shure about windows system.
Could be possible that another dependency is missing.
Maybe use tools like "dependency walker" to figure it out.