projectchrono/chrono #425: [pychrono]SCMDeformable terrain with HMMWV vehicle

shunlibiye111, 2022-11-08:
Hello everyone,I am attempting to set up a simulation with:
- hmmwv vehicle
- terrain with variable height from .bmp file
- deformable terrain with SCM model

The program runs normally,  the vehicle movement can be seen in the visualization interface and it can be seen that the terrain import is successful, but the  terrain with variable height from .bmp file cannot be visualized.

Any idea what may cause this and how to fix it?

Best Regards

20210401Yao, 2022-11-14:
nihao,woyeshi gang jiechu chrono zhege yinqin,nin keyi geige lianxi fangshi me,xiang he nin qingjiao yixie wenti.

shunlibiye111, 2022-11-14:
> nihao,woyeshi gang jiechu chrono zhege yinqin,nin keyi geige lianxi fangshi me,xiang he nin qingjiao yixie wenti.

15650580615@163.com

closed 2022-11-17 by shunlibiye111