projectchrono/chrono #431: Vehicle Module: Move JSON Keys to a Definition File

marvin-lge, 2022-12-22:
In the vehicle module, there are a several files using hardcoded keys to get values from JSON files. The problem with this approach is that if an end-user wants to change the key definitions, all these files will need to be patched. The current JSON keys use spaces in the names, which is problematic when trying to generate JSON files from tools like Matlab, which would be a possible motivation to change this. A solution would be to replace hardcoded keys with enums or macros and move the actual strings to a separate file, this would allow any third party to just patch the strings in this file if needed.

Furthermore, it is also a bit challenging to identify missing/incorrect keys at runtime, since currently a direct lookup is often used (i.e. data[group][key]), it would be useful to explicitly report in the console if a specific key is missing.

rserban, 2022-12-22:
That is not how JSON files are supposed to work. It is *not* up to the user to decide what the keys should be.  
I have worked with JSON files from Matlab and never had any issues. The issue is not with spaces in JSON key names (which are there precisely to make that easier for users).

I agree that the second item you bring up makes things difficult. The proper solution for that is defining JSON schemas. It's on my todo list, but I didn't get a chance to get to it.

closed 2022-12-22 by rserban

marvin-lge, 2022-12-22:
Thanks for your quick feedback.

I was hoping to find a convenient way align the schema with existing conventions (i.e. from an existing model or parameter database) in order to make comparison straight forward. If the schemas could be modified in a centralized place at _compile_ time, this would be convenient. Alternatively, if an "alias" could be defined that would also solve the issue. But I totally understand it is not worth the effort on your behalf.

The issue I am referring to is serializing to/from Matlab structs, it is not impossible, but it is inconvenient that the native serializer can't be used due to the spaces.
```
filename = 'C:\Git\chrono\build\bin\data\vehicle\generic\suspension\DoubleWishboneCurve.json';
text = fileread(filename);
data = jsondecode(text);
conv = jsonencode(data);
```
This is of course Matlab specific, but something that a key alias will be able to solve.

In this case a custom export process is needed to reinstate the keys, instead of just using the native json serializer from Matlab. Unless I am missing something.

Again, I'm not too bothered by this.

Thanks!