projectchrono/chrono #437: Compile chrono with future Spectra/KrylovSchurGEigsSolver.h (Modal module)

talregev, 2023-01-20:
I trying to compile chrono with  modal  module that depend on spectra library.
I compile the chrono v8.0.0 with latest spectra: v1.0.1.
It ask me for `Spectra/KrylovSchurGEigsSolver.h`. 
I search on the repo and it coming from PR that didn't merge:
https://github.com/yixuan/spectra/pull/132

My question is how you compile your chrono with future file. 
Did you checkout this pr? 

Thanks in advanced.

rserban, 2023-01-23:
Check out their `develop` branch.
That is indicated in **\*bold\*** on the Chrono installation instructions page for the Chrono::Modal module: https://api.projectchrono.org/module_modal_installation.html

closed 2023-01-23 by rserban

talregev, 2023-01-23:
@rserban `Spectra/KrylovSchurGEigsSolver.h` is not found on the develop branch. It found a PR. You should fix your documents.