projectchrono/chrono #442: Windows.h file include in ChTimer.h

markloboda, 2023-03-17:
Due to the inclusion of windows.h in ChTimer.h, the former file is exposed to the entire chrono project, achieving potentially undesirable behavior. Consequently, when chrono is run on a windows system, the windows library is directly exposed to the external file including the chrono library.

rserban, 2023-03-17:
This is an old Chrono header which should be updated. There is no reason for not simply using std::chrono on all platforms.  I'll provide a fix in a push soon.

closed 2023-03-17 by rserban

m4rr5, 2023-03-18:
And just in case anyone ever wonders if the old implementation, that uses the Windows specific QueryPerformanceCounter, is more accurate, here's an engineer from Microsoft confirming that std::chrono::highresolutionclock is actually implemented with QPC in Visual Studio versions newer than 2013: https://web.archive.org/web/20141212192132/https://connect.microsoft.com/VisualStudio/feedback/details/719443/