projectchrono/chrono #444: The vehicle can't collision with another , is there something i miss

yankaizhang20, 2023-04-07:
I create two Sedan vehicle and control one of them by keyboard. However,the vehicle which I control just go through another vehicle as like there is nothing. is there something i miss or it is what it is?

rserban, 2023-04-07:
One can define collision shapes for the vehicle chassis and enable collision for it (disabled by default). However, none of the sample vehicles and demos in Chrono illustrate that, so I'll look into adding an example.

yankaizhang20, 2023-04-07:
Thanks for your help! I have the same idea with you and did a simple test based on "demo_VEH_Sedan.cpp". **There are three differences in code and they are:**

1. define a box collision shape for the sedan chassis 
```c++
  // change.1 : add chassis collision shape
    auto chassis_mat = chrono_types::make_shared<chrono::ChMaterialSurfaceNSC>();
    my_sedan.GetChassis()->GetBody()->GetCollisionModel()->ClearModel();
    my_sedan.GetChassis()->GetBody()->GetCollisionModel()->AddBox(chassis_mat, 2, 2, 2, ChVector<>(0.0, 0.0, 0.0));
    my_sedan.GetChassis()->GetBody()->SetCollide(true);
```
2. add an obstacle with an easy box collision shape for testing the collision with sedan

```c++
// change3 :add box obstacle with collision shape
    auto ground_mat = chrono_types::make_shared<chrono::ChMaterialSurfaceNSC>();
    ground_mat->SetFriction(1.0f);
    auto obstacle = chrono_types::make_shared<chrono::ChBodyEasyBox>(5, 10, 1, 1000, true, true, ground_mat);
    my_sedan.GetSystem()->Add(obstacle);
    obstacle->SetBodyFixed(true);
    obstacle->SetPos(chrono::ChVector<>(-10, 0, 0));
    obstacle->GetVisualShape(0)->SetTexture(chrono::GetChronoDataFile("textures/concrete.jpg"));
    obstacle->SetCollide(true);
```

3. add some particles same as in "demo_IRR_assets.cpp" for testing the physics behavior of "Terrain"

```c++
// change2 : add particle same as in demo_IRR_assets.cpp
    auto particles    = chrono_types::make_shared<ChParticleCloud>();
    auto particle_mat = chrono_types::make_shared<ChMaterialSurfaceNSC>();
    particles->GetCollisionModel()->ClearModel();
    particles->GetCollisionModel()->AddSphere(particle_mat, 0.05);
    particles->GetCollisionModel()->BuildModel();
    particles->SetCollide(true);

    // Create the random particles
    for (int np = 0; np < 100; ++np)
        particles->AddParticle(ChCoordsys<>(ChVector<>(ChRandom() - 2, ChRandom() + 2, 3)));

    // Mass and inertia properties.
    // This will be shared among all particles in the ChParticleCloud.
    particles->SetMass(0.1);
    particles->SetInertiaXX(ChVector<>(0.001, 0.001, 0.001));

    // Do not forget to add the particle cluster to the system:
    my_sedan.GetSystem()->Add(particles);

    //  ==Asset== Attach a 'sphere' shape asset.. it will be used as a sample
    // shape to display all particles when rendering in 3D!
    auto sphereparticle                     = chrono_types::make_shared<ChSphereShape>();
    sphereparticle->GetSphereGeometry().rad = 0.05;
    particles->AddVisualShape(sphereparticle);
```

there are two conclusions:
- it can't take collision check by add collision shape for chassis in my way, as the following picture showing.
- the particles just go through the terrain and have not collision behavior as the MBS demo.
[image]

Are there some errors when i add collision shape for Sedan chassis? Why the particles can’t collision with terrain and how to do it?

rserban, 2023-04-07:
1. You did not invoke BuildModel() on the collision model you added to the Sedan chassis. Note that there is a more formal way of adding collision geometry to a vehicle chassis (see below)
2. You created the particle cloud with an "NSC" contact material, while the system you place it in is of type "SMC" (the two contact formulations cannot be mixed).  

I just pushed a demo that illustrates the use of collision geometry on a vehicle chassis.  See demo_VEH_HMMWV_Rollover. Look also at the implementation of HMMWV_Chassis; I used a collision shape given as a convex hull (from a simplified mesh), but you could define a collision model composed of primitives (that requires a bit of extra work to properly match the collision model with the visual model)

yankaizhang20, 2023-04-10:
Got it. Thanks so much!

closed 2023-04-14 by rserban