projectchrono/chrono #451: Step files cannot be visualized, demo_CAS_stepfile

AramNasser, 2023-05-23:
In the demo_CAS_stepfile.cpp, I am unable to visualize the step files I load, yet it is visible among the object list: "sys.Get_bodylist();"
i.e. the file is loaded but cannot show in the GUI.

Can you help, please? Thank you in advance

One note: while doing the compilation, I got two errors in the ChCascadeIrrMeshTools.h library:
class "Poly_Triangulation" has no member "Nodes"
class "Poly_Triangulation" has no member "Normals"
I commented out the "static void fillIrrlichtMeshFromCascadeFace" function (the function that calls the two objects) declaration and its call. could that be the problem? if so, what do you advise?

Looking forward to hearing from you 
Sincerely

rserban, 2023-05-23:
Of course, with that function commented out you will not get any visualization.
What version of OCC do you use? Note that Chrono can only work with version 7.4.0.  And in that version, both functions Poly_Triangulation::Nodes() and Poly_Triangulation::Normals() are defined.

AramNasser, 2023-05-23:
Thank you for your prompt reply

Yes, I use opencascade-7.7.0. I will try to recompile and link to the OCC 7.4.0 and hope the problem will be solved.

Thanks again and have a nice day

AramNasser, 2023-06-07:
Hi again,

I have redone the installation with OCC 7.4.0 but am still unable to visualize the loaded step files, nor ChBodyEasySphere or ChBodyEasyBox when their add function is called from a GUI button's event function. I can visualize them only when called in the Main function before
vis->AttachSystem(&sys);
vis->SetWindowSize(800, 600);
vis->SetWindowTitle("ChCascade");
vis->Initialize();

....and the other settings

Please note that when I do "sys->Clear();" from the button's event function the objects disappear, meaning the reference to the sys is working well. Do I miss anything to show the added objects?

What do you advise in this case?

rserban, 2023-06-07:
When you add a physics item (such as a body) to the system **_after_** the associated visual system was initialized, you must explicitly inform the visual system of the new assets (the item's visual model). You do this by explicitly invoking the function `ChVisualSystem::BindItem()`. Note that all visual models defined at the time you initialize the visual system are automatically bound (so no need for any explicit call in that case).

AramNasser, 2023-06-07:
Great, thanks

However, now the added elements just float. How can I add the physical attributes to them?
For example, the same objects fall down when I call them from the Main

rserban, 2023-06-07:
This is likely some bug in your code.   Not sure what you mean by "adding the physical attributes".  Are you sure you do not set them as "fixed"?

AramNasser, 2023-06-07:
Thanks for your prompt reply

Yes, I am sure they are not fixed. They fall when they are called from the main function....
About "adding the physical attributes", I meant to make them fall as a physical object not stay afloat. 

Should you need any further details, please do not hesitate to ask. Thank you in advance.

AramNasser, 2023-06-23:
Hi again @rserban

Please, let me clarify the issue:

I am calling the function "AddFallingItems(sys)" that exists in the demo: "demo_MBS_collisionNSC.cpp" from a button. Now the items are visible and everything is OK (after installing the right OCC 7.4.0  and binding the elements.)

However, when I call the function from the main, the items fall as they do in the demo but when I call it from the button, they do not fall unless they touch the falling items loaded from the main. When the items loaded from the main are no longer falling, i.e. they touch the bottom of the container "AddContainer", the newly added items stay at their positions and do not fall (this also applies to the items loaded using the LoadStepModel(....))

I tried to update the ChSystemNSC sys (sys.Update()) but it seems it has no effect!

Have I missed anything? 

Thank you in advance

rserban, 2023-06-26:
There must be a mistake somewhere in your code. A very simple modification of demo_MBS_collisionNSC where the falling objects are created at some later point in the simulation, using something like:

```
    bool create = true;
    while (vis->Run()) {
        if (create && sys.GetChTime() > 3) {
            AddFallingItems(sys, vis);
            create = false;
        }

        vis->BeginScene();
        vis->Render();
        vis->EndScene();

        sys.DoStepDynamics(step_size);

        rt.Spin(step_size);
    }
```
postponing the generation of falling objects and with an appropriately modified `AddFallingItems()` to pass the visualization system so that the newly created bodies can be bound to the vis system, works exactly as expected.

AramNasser, 2023-06-27:
Thank you so much indeed for your time and effort

You are right, the problem was in my code as I sat "sys.SetUseSleeping(true);" to test something and forgot to reset it back to false, or delete it!

In case someone faced the final issue I mention here, just turn this feature off!  
    /// Turn on this feature to let the system put to sleep the bodies whose
    /// motion has almost come to a rest. This feature will allow faster simulation
    /// of large scenarios for real-time purposes, but it will affect the precision!
    /// This functionality can be turned off selectively for specific ChBodies.
    void SetUseSleeping(bool ms) { use_sleeping = ms; }

Thanks again and have a great day :)

closed 2023-06-27 by AramNasser